supermerill/SuperSlicer #3253: Print was finished before to actual print was printed (2.5.59 - Arachne)

tloesch, 2022-09-22:
### What happened?

The Printer finished the gcode without any errors.
The Print was somehow not finished.

I had already a strange observation while printing:
Layer Height didnt matched the Layer Count (observed in Mainsail)

**Requested Height/Count:**  11.60 mm / Layer 88
**Result Height/Count:** ~ 5.04 mm / Layer 47

Picture of the expected outcome:
[image]

Picture of the result:
[image]

### Project file & How to reproduce

[layerheightproblemproject.zip](https://github.com/supermerill/SuperSlicer/files/9624457/layerheightproblemproject.zip)

I attached the Project and also my configuration bundle.

**Basic Informations:**
- Arachne is enabled
- On the Part the Base Layer Height is set to 0.15 mm

### Version

2.5.59.0 (from Pre-Release)

### Operating system

Windows 10 - Version 21H2 (Build 19044.1949)

### Printer model

LDO Voron 2.4 300mm with Stealthburner, Galileo and Dragon HF (0.4 mm Nozzle)

neophyl, 2022-09-22:
To me that just looks like you have had an extruder clogged issue while printing.  At about 1mm up you can see what looks like missing filament around the outer perimeter.  If the extrusion was having difficultly at that point and then recovered for a bit then there is every reason to suspect that it simply failed to feed at a later point resulting in the missing top of the model.  The printer 'thinks' its printing so carries on until it has finished the gcode.  

As for the slicing layer height/count differences, that is normal when supports are being used.  They have their own heights, not tied to the model.  Turn supports off and slice and the layer count is 72, which for a first layer of 0.24 and then 71 layers of 0.16 equals 11.6mm which is the height of your object, or as close as it cant get to 11.65.  When supports are turned back on the layer count increases due to layers that only have support.  So normal function.

Btw you may have set 0.15 as your object layer height and the first layer as 0.25 but your printer settings has a z full step configured of 0.04mm which means 0.25 and 0.15 are not achievable given the constraints you are giving the slicer.  So its using 0.24 and 0.16 which is the closest it can get to your 'wanted' values.

tloesch, 2022-09-22:
Thanks for clearing up the layer height / layer count topic for me.
Also thanks for the information about the wrongly configured z-full-step.

I will check for a clocked hotend and fix the z-full-step config.
After that i will report if the issue still exists.
If not i will close this issue.

tloesch, 2022-09-26:
Issue was a clocked hotend. Thanks @neophyl 
Sorry for the issue. Next time i will make more checks before opening an issue.

closed 2022-09-26 by tloesch