supermerill/SuperSlicer #3256: Snug supports - creating raft for supports that don't exist?

nirurin, 2022-09-22:
### What happened?

Using snug supports, there are patches of raft being made for supports that aren't being created. If I increase the 'support angle' a bit to get more supports added, the rafts will gradually get supports added. It's like the 'raft' part of the support setting is set to a different tolerance than the actual 'support' part. 

### Project file & How to reproduce

[image]

### Version

2.5.59

### Operating system

windows 11

neophyl, 2022-09-22:
It is. It’s the same in PS too. If the support area is too small the actual support gets registered but not generated as there’s a setting for the spacing. It’s the same way that if you have your xy separation too high then it can cause supports to be deleted which gives a non contiguous support structure (another bug for supports). Unfortunately the support expansion still gets applied.  Which as the default is 3mm gives you the islands. I normally use an expansion of zero anyway on my profiles so I don’t see them.

nirurin, 2022-09-22:
Interesting. I do like snug supports for some things, but it does also seem to produce some supports which are just so thin that they can't even support themselves. Shame they don't have an option for some kind of 'tower' or 'tree' like in cura, where the support can have a wider base that goes up to the snug point at the top.