supermerill/SuperSlicer #3257: 'Supporting Dense Layer' infill not working under smaller areas of top-skin

nirurin, 2022-09-23:
### What happened?

Printing a model which has some slopes that have infill under them, but the outer perimeters of the slopes are caving in because there's no support under them. I don't have very dense infill, but that's never been a problem because I use the 'supporting dense layer' option. This option works for the main top layer, so it's definitely 'on', it's just not working correctly (or there's another setting I'm not finding to alter it?)

I checked the same model with the same settings on prusaslicer, and it's putting the dense layer under that section as it should, so it's not a problem with the model. 

### Project file & How to reproduce

[image]

As you can see, works fine in prusaslicer (left). 

[Valorant-Chamber-SniperPlating_TopRail_batterycover_lipoly.zip](https://github.com/supermerill/SuperSlicer/files/9630889/Valorant-Chamber-SniperPlating_TopRail_batterycover_lipoly.zip)

### Version

2.5.59

### Operating system

windows 11

### Printer model

Ender5+

neophyl, 2022-09-23:
That is NOT supporting dense layer.  That is solid infill.  And usually its very undesirable as it causes excess heat in those areas where it can cause all sorts of issues.  Its why there have been MANY issues both here and over on the prusa forum asking for it to be fixed.  As well as excess heat it shakes the printer around and increases print time and in many cases doesn't really do anything anyway.
Super Slicer though put in a fix for it which is why its not adding it.  Technically it is controlled by the setting 'Ensure Vertical Shell Thickness'  and in PS that is the only thing you have to effect it.  As mentioned though PS adds it most times even if that setting is turned OFF and you dont want it.  In PS the usual work around to get rid of it is to increase perimeter count.  For functional props as it appears you are printing that is generally a good idea anyway.  Increasing perimeters results in a stronger part than increasing infill %.  You can usually reduce infill too and get a faster print too.

In SuSi you may notice a setting next to Ensure Vertical shell thickness, called NO Solid Infill over.  By changing that value you can control if it is added or not.  SuSi still does things differently though as it will often add a thicker gapfill line.  

I tried to slice your project but I'm getting an error of 'No Extrusions were generated for objects'.  I had to replace your printer profile to get it to slice.  

The supporting dense layer setting (which PS doesn't have ) is located in Infill>Reducing Print Time.  Your profile doesn't have it  enabled.  When it is enabled it add a 50% bridging layer under any top layers to give then some initial support.  This helps when you are using lower infill percentages.  There are various options to change how it applies the 50% supporting dense layer.  Experiment with the settings until you get an understanding of what they actually do and how they work.

nirurin, 2022-09-23:
> That is NOT supporting dense layer. That is solid infill. And usually its very undesirable as it causes excess heat in those areas where it can cause all sorts of issues. Its why there have been [...]

I was going to say, I already checked and made sure I had 'solid infill' turned off. I was actually testing turning it -on- to try and see if that helped, but it did something totally unrelated (as I expected). But if you're saying this is something prusaslicer does all by itself then fine, I had no control over it. Just funny that it is doing the thing I -wanted- superslicer to do with the 'supporting layer' option haha. 

> In PS the usual work around to get rid of it is to increase perimeter count. For functional props as it appears you are printing that is generally a good idea anyway. Increasing perimeters results [...]

I already have 3 perimeters with a 0.6 nozzle, which is pretty thick. I could add more, but I've never needed anywhere near that much in the past (supporting dense layer over low infill% has always worked for me in previous versions)

> 
> In SuSi you may notice a setting next to Ensure Vertical shell thickness, called NO Solid Infill over. By changing that value you can control if it is added or not. SuSi still does things differently though as it will often add a thicker gapfill line.
> 

I may try this for now, as it would at least be a solution. I'll have a play with it. 

> I tried to slice your project but I'm getting an error of 'No Extrusions were generated for objects'. I had to replace your printer profile to get it to slice.

Profile works fine for me (barring this current infill issue) so not sure why the profile isn't working for you. May be cos I'm working from a nightly release? (I have no choice, the main release is bugged to hell for me)

> 
> The supporting dense layer setting (which PS doesn't have ) is located in Infill>Reducing Print Time. Your profile doesn't have it enabled. 
[image]

This setting? I have it turned on. Unless there's another setting there I'm not seeing? 

> When it is enabled it add a 50% bridging layer under any top layers to give then some initial support. This helps when you are using lower infill percentages. There are various options to change how [...]

If its the setting I have turned on, then it's doing nothing. If there's a missing setting, how do I get it back? Could you show me what the menu is meant to look like? These RC/nightly editions have had a bunch of weird UI bugs. I lost the arachne dropdown for a while too.

neophyl, 2022-09-23:
In the project I loaded from you and opened in the non nightly 2.5.59.0 running on Windows 10 then that setting is UNticked.  When I ticked it and sliced it added the supporting dense layer as it should.  

I suspect something screwy is going on with your profiles and that half the issues you are having isnt because the software is broken but down to something in your profiles/projects.  
I hope you have been doing regular backups of them using the  Export Config bundle option.

nirurin, 2022-09-23:
This profile is only one day old, as I had to completely wipe my old
profile data and recreate it while trying to fix the major crashing bug the
main release currently has.

I can create it again from scratch easily enough, but the setting is
definitely turned on according to my ui (as you can see).

I can only assume the bug happened because I opened the same new profile in
both 2.4.58 and 2.5.58 and it didn't like it for some reason? But there's
no way to prevent that as all superslicer versions get their configs from
the same user data folder.

I do have a backup, but it's from the older version I was using up until a
week or so ago, pre arachne, so it'll be just as likely to cause issues.

I don't really see how me making the profile (which is just raw text at the
end of the day) could cause a bug anyway. I only did exactly what I was
meant to do. Filled in boxes with numbers and ticked some boxes. Saying
that me doing that is "screwy" seems a bit harsh. Is that is all it takes
to break the app, then the software is broken.

nirurin, 2022-09-23:
Could you maybe export your version of my settings, with that option ticked? Maybe that will work... *shrug* worth a try!

neophyl, 2022-09-23:
I did figure out what was causing the no extrusion error .    In Print Settings>Perimeters & Shell>Arachne the Minimum Feature Size was set to 25mm.  Not surprising it wouldnt slice.

You definitely have something screwy going on.  I don't mean you are doing anything wrong, I just mean the stored data is out of whack.

Normally on non stable releases I use a run switch to define the data path , that each version of SuSi has its own data (usually stored in the same folder with the exe).  SO yeah I have to run through the wizard initially for each new version but then I just import my saved config bundle and I get all my normal stuff back.  That way they dont interfere with each other.

Then on projects I save them with the SuSi or PS version tacked onto the file name.  Benchy_PS2.5_3mf or Benchy_SuSi2.25590.3mf for example.  Saves a world of hurt.

I can give you an example I saw in action on just how profiles can mess things up even if they are 'just text'.  This applies to both PS and SuSi.  Once upon a time :)  a member on the prusa forum installed a Printer profile from the interwebz.  All looked good.  They couldnt figure out though why when they had settings on the print section ticked it wasnt respecting that tick.

Turns out the profile for the Printer had items from the Print Settings mixed in with it.  So when the setting was ticked under Print Settings=On the same named setting in the Printer Settings Section was =Off.  As the Printer Settings couldnt actually change the instance in its section and as that was later in the config the end result was that the option always evaluated as OFF.

This was all down to a printer manufacturer putting a 'screwy' profile on to the web and of course all those people downloaded it.  
I am not saying this will be exactly the same, Im just saying that something is obviously not right and its just an example of how simple text files can and do mess things up (sometimes).

nirurin, 2022-09-23:
> I did figure out what was causing the no extrusion error . In Print Settings>Perimeters & Shell>Arachne the Minimum Feature Size was set to 25mm. Not surprising it wouldnt slice.

I checked, and my minimum feature size is set to 25% not 25mm. 
I just opened my 3mf file on my superslicer, and it shows that 'supporting dense' option as being ticked. 
Hard to figure out whats going on.

nirurin, 2022-09-23:
Tested now using just a base install of superslicer, with no custom profile settings at all (all deleted). 

Dense supporting layer only prints underneath 'solid infill'. If there's no solid infill, there's also no supporting infill.