supermerill/SuperSlicer #3259: Raft Contact Z Distance Not Affecting Slicing

ruredi87, 2022-09-23:
### What happened?

Whatever I set raft contact distance does not change anything. It has fixed gap and I can't change it 

### Project file & How to reproduce

[raft bug.zip](https://github.com/supermerill/SuperSlicer/files/9637089/raft.bug.zip)
[image]
[image]

### Version

SuperSlicer_2.5.59.0_2022-09-13

### Operating system

Windows 11

### Printer model

Ender 3 v2

neophyl, 2022-09-24:
Mmm your own pictures show a change.  The top picture has a gap between the bridge infill (light blue, looks like its printing at the default nozzle diameter=layer height so thicker than the normal layer height.  The bridge infill is the bottom of your print layers not the perimeters in this case due to your settings.

Your second picture shows the bridge infill touching the raft so no gap.  Clearly a change.  Seems to all be working normally.

ruredi87, 2022-09-24:
> Mmm your own pictures show a change. The top picture has a gap between the bridge infill (light blue, looks like its printing at the default nozzle diameter=layer height so thicker than the normal [...]

Thank you I could not see that

closed 2022-09-24 by ruredi87