supermerill/SuperSlicer #3262: Version check checks PRUSA not SS.

inspectionsbybob, 2022-09-24:
### Version
2.4.58.4

Win10

### Behavior
Got Notified of a New Release when I opened SS.
Clicked on LInk and it took me to PRUSA SLICER 

There IS a new SS release, 2.4.58.5   but that is not where it took me...

_

neophyl, 2022-09-24:
Did you not read the release notes for the latest version ?  It mentions fixing the hard coded PRUSA download page.

inspectionsbybob, 2022-09-24:
Sorry, I went through the read me and the change log but there was no mention of that that ai saw of the first pass...

neophyl, 2022-09-24:
I think it’s this one 
change download page from hardcoded prusa 
At least that’s the way I read it.

inspectionsbybob, 2022-09-25:
With a description like that, no wonder I missed it.  Anyway, so long as the development team knows, close this one as a duplicate.

inspectionsbybob, 2022-09-25:
Duplicate

closed 2022-09-25 by inspectionsbybob