supermerill/SuperSlicer #3263: Sparse fill pattern modifier for parts are not retained over save files

feketeimre, 2022-09-24:
### Version
2.4.58.5

### Operating system type + version
Windows 10

### Behavior
* _Describe the problem_
Sparse fill pattern modifier for parts are not retained over save files
* _Steps needed to reproduce the problem_
Create new project, Add shape/Box, Right-click on box, select Infill option to add modifier, save project, load the just saved 3mf file.
* _Expected Results_
The part from the loaded file should have the sparse fill pattern modifier
* _Actual Results_
The part from the loaded file does not have a parse fill pattern modifier, only infill percent. Infill pattern type set for the part is lost when saved.
I looked into the 3mf file and the infill type data is not even saved in the Metadata\Slic3r_model.config file
[image]

_Is this a new feature request?_
No

NameOfTheDragon, 2022-09-24:
FYI, this is working correctly in 2.5.59.0

feketeimre, 2023-12-31:
@NameOfTheDragon
Thanks for pointing it out!

closed 2023-12-31 by feketeimre