supermerill/SuperSlicer #3264: Bridge Infill Causes Issues Over Raft

ruredi87, 2022-09-24:
### What happened?

Superslicer's bridge infill causes issues when using raft. I am uplading 3 pictures with number on them. Every setting's raft z distance is 0, so there should not be any gap between object and raft interface layer. 

Number 1 :"Nozzle diameter" is set with bridge flow baseline.(The problem is there is no distance between bridge infill and raft as expected. but if you look at number 1 picture you can see there is gap between walls and raft layer)
[image]

Number 2:"Layer height" is set with bridge flow baseline.(The problem is bridge line width is too small around 0.12mm when using 0.3mm nozzle with 0.25mm line width.Because of this, object does not stick to raft layer and causing problem.)

[image]

Number 3:"Keep current flow" is set with bridge flow baseline.(The problem is there is no distance between bridge infill and raft as expected. but if you look at number 3 picture you can see there is gap between walls and raft layer)
[image]

I could not work around this. If we could disable the bridge infill setting probably it can help the issue.

### Project file & How to reproduce

[bmg project.zip](https://github.com/supermerill/SuperSlicer/files/9639669/bmg.project.zip)

### Version

Build: SuperSlicer_2.5.59.0_2022-09-13

### Operating system

Windows 11

### Printer model

Ender 3 v2

neophyl, 2022-09-25:
Set the bridge infill to use layer height option. It’s not idea as the width is also set to the layer height but it does work around it.

ruredi87, 2022-09-25:
> 

Its causing extremly low layer widht so parts does not stick to raft. I am trying to print with abs, pla might stick but still it is a issue

Fabface, 2022-09-26:
Try setting this to "from plane" - that does what you want for me
[image]

ruredi87, 2022-09-26:
> Try setting this to "from plane" - that does what you want for me [image]
[image]

Thank you for your reply. I tried your setting than I set raft z distance to 0.1mm as you see in the image there is 0.1mm gap between walls and raft but there is no gap between bridge and raft. This is the problem. Why superslicer using bridge with raft I dont understand. Problem not solved

Fabface, 2022-09-26:
Hmmm thats strange, for me it works
[image]
For me its rather the opposite, I cant manage to have the overhang layer sitting on the raft, its always as gap there, no matter what I do.
[image]

ruredi87, 2022-09-26:
> Hmmm thats strange, for me it works [image] For me its rather the opposite, I cant manage to have the overhang layer sitting on the raft, its always as gap there, no matter what I do. [image]

Which version are you using

Fabface, 2022-09-26:
I use 2.5.59.0

ruredi87, 2022-09-28:
I noticed when I select support z offset to soluble superslicer gives error about empty layers but there is actually raft under the object maybe this is causing the issue. I think superslicer does not recognize raft as layers so it is trying to add bridge. Also it would be good to add disable detect bridges like in prusa slicer.

[image]

git2212, 2024-03-25:
This is a rather old bug report however, this odd behavior is still present in 2.5.59.8

Choosing "layer height" for bridge infill does not actually provide a workaround. The problem is that superslicer will use a bridge infill instead of a solid infill for the first layer right on top of the raft... see pic below
[image]

whatever settings we tune for bridge infill may work at the top of a part but definitely will not work for the first layer... consequently printing on a raft is not really working. 

Is there a planned fix for this issue?... Printing on raft is important for materials which warp or parts with certain size and geometries.