supermerill/SuperSlicer #3269: M109 / M190 too far from M104 / M140

Shine753, 2022-09-25:
When the GCode is generated, there are tons of things between M104 / M140 and M109 / M190.
So if you home the printing head into your start gcode, the filament continue to flow at home position until the temperatures are reached. That can lead to a blob at the nozzle exit. This also leads to problems if you do some filament purge after homing in the start gcode : the purge can start before temperatures are reached.

Is there a reason explaining why these commands are not coupled together ?
It would be great to have them coupled, and not spaced as it is now, so nothing happens until the temperatures are reached. Or having an option to do so.

diggit, 2022-11-02:
Can you share your gcode?
Which superslicer version are you using?
Which gcode flavor are you using?
Do you have Printer->Custom G-code-> Start g-code filled with something?

Project file would help.

Following bug report template would tell you, what is necessary to provide.

after commenting out my custom printer startup gcode, I get:
(klipper flavor)
```gcode
M73 P0 R6
;TYPE:Custom
; custom gcode: start_gcode
;SET_FAN_SPEED FAN=Nevermore SPEED=1 ;start nevermore
;M109 S0 ;workaround to disable M109 automatic inclusion after start macro
;M190 S110 ; force bed to heat-up first
;print_start EXTRUDER=245 BED=110 CHAMBER=50
;M109 S245; wait for hotend temperature to settle
;PRIME_NOZZLE
; custom gcode end: start_gcode
M107
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
; custom gcode: start_filament_gcode
; Filament specific G-code start
SET_PRESSURE_ADVANCE ADVANCE=0.03 SMOOTH_TIME=0.01
; Filament specific G-code end
; custom gcode end: start_filament_gcode
;_TOOLCHANGE 0
M104 S245 ; set temperature 
M109 S245 ; set temperature and wait for it to be reached
M190 S110 ; set bed temperature and wait for it to be reached
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
; custom gcode: before_layer_gcode
;BEFORE_LAYER_CHANGE
;0.2

; custom gcode end: before_layer_gcode

G1 Z0.2 F30000
; custom gcode: layer_gcode
;AFTER_LAYER_CHANGE
;0.2
; custom gcode end: layer_gcode

G92 E0
M73 P4 R6
G1 E-1.5 F3600
...
```

Shine753, 2022-12-21:
Using SuperSlicer 2.4.58.5, with Marlin 2 GCode,
Here is the generated GCode.
As you can see, there is a problem when the purging procedure starts, the parts are not yet heated.

```
; generated by SuperSlicer 2.4 on 2022-12-21 at 12:19:36 UTC

; 

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.40mm
; infill extrusion width = 0.46mm
; solid infill extrusion width = 0.48mm
; top infill extrusion width = 0.46mm
; first layer extrusion width = 0.40mm

; object:{"name":"Forme-Box","id":"Forme-Box id:0 copy 0","object_center":[115.500000,115.500000,0.000000],"boundingbox_center":[115.500000,115.500000,11.250000],"boundingbox_size":[22.500000,22.500000,22.500000]}
; Total objects to print: 1
M486 T1
; plater:{"center":[115.500000,115.500000,0.000000],"boundingbox_center":[115.500000,115.500000,11.250000],"boundingbox_size":[22.500000,22.500000,22.500000]}

M73 P0 R21
M140 S65 ; set bed temperature
M104 S230 ; set temperature
;TYPE:Custom
;=========================================== This starts before nozzle and bed reaches the targeted temperature
G28 ; homing
;M117 Purging Extruder... ; ------------------------------------ Extruder purge
;G1 Z1.0 F3000 ; lift Z
;G1 X0.2 Y25 Z0.2 F3000.0 ; move to start-line position
G92 E0 ; reset the extruder
;G1 X0.2 Y150.0 F1000.0 E15 ; draw 1st line
;G1 X0.7 Y150.0 F3000.0 ; move to side 0.5mm
;G1 X0.7 Y50 F1000.0 E30 ; draw 2nd line
;G1 X0.7 Y25 F5000.0 ; quick wipe
M117 Printing...
;===========================================
M107
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament gcode
M92 E98 ; ------------------------------------ Filament EStep
M500 ; ------------------------------------ Save
;_TOOLCHANGE 0
M109 S230 ; set temperature and wait for it to be reached
M190 S65 ; set bed temperature and wait for it to be reached
M107
;LAYER_CHANGE
;Z:0.28
;HEIGHT:0.28
;BEFORE_LAYER_CHANGE
;0.28
[...]
```

diggit, 2022-12-21:
SS by itself, heats up printer **after** custom `Start g-code`, so it's your responsibility to insert appropriate g-code, which sets temperature and waits for settling, if you need it there.

Here is my `Start g-code`:
```gcode
M109 S0 ;workaround to disable M109 automatic inclusion after start macro
M190 S[first_layer_bed_temperature] ; force bed to heat-up first
M104 S{first_layer_temperature[0]}; start extruder heat up
PRINT_START; homes printer etc.
M109 S{first_layer_temperature[0]}; wait for hot-end temperature to settle; possibly redundant to print_start
PRIME_NOZZLE
```
`PRINT_START` and `PRIME_NOZZLE` are just macros which get replaced by some other gcode, contents not relevant here. You can adapt it to your needs.

Shine753, 2022-12-22:
Thank you, that works like a charm.

closed 2022-12-22 by Shine753