supermerill/SuperSlicer #3274: Internal bridge over bottom skin layers

PhilBaz, 2022-09-26:
### What happened?

Internal bridge where it shouldn't be. On top of bottom skin (solid infill )layers.

### Project file & How to reproduce

[image]
[Bridge over skin.zip](https://github.com/supermerill/SuperSlicer/files/9648860/Bridge.over.skin.zip)

### Version

2.4.58.5

### Operating system

win 10

### Printer model

Klipper TronXY X2 custom

kpdcompany, 2022-09-29:
Can we PLEASE get a setting to turn off or threshold this Interior Bridge Fill?

neophyl, 2022-09-30:
For the original posters issue then you have Print Settings>Infill>Reducing Print Times>Supporting Dense layer Enabled (with the automatic option).  Untick that and it no longer adds a 50% supporting layer for layers above that you have pictured in your screen shot.

kpdcompany I don't know if the same setting is what you are looking for as you haven't provided a project to examine.

neophyl, 2022-09-30:
The original posters attached project DOES have that setting enabled when opened. You haven’t attached any project to this issue that I can see.

kpdcompany, 2022-09-30:
Sorry, I though this was a reply to my issue.  Mine does not have it
checked and it is still there.

https://github.com/supermerill/SuperSlicer/issues/3295

Travis Kemp
President at Kemp Prototyping & Design Co. LLC
<https://g.page/KPDCompany/review?rc>
<https://www.instagram.com/kpdprototypedesign>
<https://www.linkedin.com/company/72652573>
[image: T:]  513-460-7366 [image: E:]  ***@***.***
www.KPDCompany.com <http://www.kpdcompany.com/>

PhilBaz, 2022-09-30:
@neophyl 

Yes, you are right.  Settings>Infill>Reducing Print Times>Supporting Dense layer disabled removes it.

But then this is definitely a bug, as should only apply under top solid infill.

Settings>Infill>Anchor solid infill by X, also seems to have an impact.

PhilBaz, 2022-09-30:
@neophyl 

But there is no solid infill above it?

kpdcompany, 2022-10-03:
This is absolutely a bug.  There is NO way to actually turn off this infill bridge.  Supporting dense layer is off and it still does it for me no matter what.

kpdcompany, 2022-10-03:
Sorry but didn't you just say it could be broke?  I was replying to him to let him know its quite possible its not working correctly.  And it is absolutely a bug that the "Supporting dense layer" does not turn off??

Not sure how that is a feature request, only thing I really requested was a gap setting to actually effect all solid layers and a gap setting for the top solid.  

The rest is the real issue, not being able to turn this supporting layer off is a bug I thought according to  your response?

PhilBaz, 2022-10-03:
For the record, "Supporting dense layer" off does turn off this feature for me. And I understand why its there, and it is impacted by  Settings>Infill>Anchor solid infill by X, that creates a small band of overhanging skin, which then initiates the supporting layer. 

Its not super intuitive, since it affects the whole layer and not just the section under the skin band. But it is consistent, and makes sense.

closed 2022-10-03 by PhilBaz