supermerill/SuperSlicer #3275: Compare Presets: Move Setting Values From One Preset to Another

Garryb123, 2022-09-26:
**Is your feature request related to a problem? Please describe.**
I waste time updating preset values after comparing presets. Obviously this could be done much more efficiently.

**Describe the solution you'd like**
When using the compare tool I would love to be able to simply click an arrow button to move setting values from one preset to the other.

**Describe how it would work**
A single button click would move values from one preset to the other.

**Describe alternatives you've considered**
The only way to update presets is to enter any preset and find the setting and change the value.

neophyl, 2022-09-26:
Duplicate of https://github.com/supermerill/SuperSlicer/issues/2524 I think.

Garryb123, 2022-09-26:
> Duplicate of #2524 I think.

Thank you; that request seems similar. Sorry for duplicating.

#2717 seems to be exactly what I want.

I don't fully understand supermerill's proposal:

what I propose:

when you click on the name, it will send you to the setting, like for the search.
when you click on a value (whatever the one), your current setting is switched to this one
refresh without resetting the selected presets when you save your current preset.

My idea is for all matching/updating to be done when in Compare Presets. When comparing the settings, you see the differences, and then move the one you want to where you want it just by clicking an arrow button that points the direction you like. Very simple; (at least in theory - I don't know if programming would be easy).