supermerill/SuperSlicer #3281: cannot load monotonic filled preset

gratefulfrog, 2022-09-27:
### What happened?

I save a print settings preset with:
infill solid : montonic filled

I verifiy in the file on disc that:
solid_fill_pattern = monotonicgapfill

I close Superslicer and reopen it.

The solid infill pattern is now : Rectilinier (filled)
and the print settings is marked as (modified)

I can save it as many times as I want, onn reloading it always goes back to Rectilinear (filled)

This creates wrong gcode, obviously....

Thanks for your help

### Project file & How to reproduce

I save a print settings preset with:
infill solid : montonic filled

I verifiy in the file on disc that:
solid_fill_pattern = monotonicgapfill

I close Superslicer and reopen it.

The solid infill pattern is now : Rectilinier (filled)
and the print settings is marked as (modified)

I can save it as many times as I want, onn reloading it always goes back to Rectilinear (filled)

This creates wrong gcode, obviously....

Thanks for your help

### Version

2.4.58.3

### Operating system

ubuntu 18.04

### Printer model

voron 2.4 afterburner

gratefulfrog, 2022-09-27:
[ss.zip](https://github.com/supermerill/SuperSlicer/files/9653892/ss.zip)

the project and print settings file

nucelus, 2023-06-02:
I have the same issue, it happens during loading a saved profile and when opening a saved project. 

Solid Infill: Rectilinear (filled) is always selected when loading up superslicer with Monotonic (filled) saved previously.

Flipqy777, 2024-04-18:
I have the same issue too. saving profiles with "Monotonic (filled)" top pattern always reverts to "Rectilinear (filled)" when SuperSlicer is closed then reopened.

closed 2024-06-05 by supermerill