supermerill/SuperSlicer #3282: Need function infill one direction for Print Glass ?

Witawat, 2022-09-27:
Need function infill one direction for Print Glass ?

https://www.printables.com/model/15310-how-to-print-glass/comments

jhlchu, 2022-09-27:
Set `Print Settings > Infill > Angle > Increment = 90`. Close issue.

neophyl, 2022-09-27:
Or use aligned rectilinear with zero top and bottom layers.

Witawat, 2022-09-28:
> Set `Print Settings > Infill > Angle > Increment = 90`. Close issue.

not work
[image]
[image]
[image]

jhlchu, 2022-09-28:
Re-read what I wrote. You changed `Fill Angle` not `Increment Angle`.

Witawat, 2022-09-28:
> Increment Angle

[image]

thank you work

closed 2022-09-28 by Witawat