supermerill/SuperSlicer #3292: Slice missng vertical print is not equal

drveni, 2022-09-29:
### What happened?

I found this bug earlier. I do not understand why. this error always appears when I have a gaping hole on an object on hole-side. then the beginning of the lower side of the hole and the end of the upper side always has some transition that can be seen in the print and is always generated in the slicer. maybe I have some settings wrong, but I don't believe it.. this has been going on for a long time and now it's getting annoying. simply that transition is visible and is always generated in the slicer. these objects have vertical FLAT sides. but when laying where there is a hole, the crack always has an error along the entire length (round object).
[image]
[image]
[image]

### Project file & How to reproduce

it doesn't matter which object it is, the same problem and error occurs on all of them

### Version

2.5.59

### Operating system

mac m1

### Printer model

ender 3v2 custom

jhlchu, 2022-09-30:
Attach project file

jhlchu, 2022-09-30:
Shot in the dark, it's `Print Settings > Slicing > Vertical Hole Compensation` set `XY Compensation = 0` should fix. See related #3197.

drveni, 2022-09-30:
> Attach project file

I don't have Z-axis compensation. I only have XY, you can see it in the picture

I say, it's not a problem in this STL, it's a problem in everyone who has it/holes.

[image]
[odrezan-carriage.stl.zip](https://github.com/supermerill/SuperSlicer/files/9686095/odrezan-carriage.stl.zip)

neophyl, 2022-09-30:
There is nothing wrong with the stl.  
[image]
See no lines.  Your own picture shows you have XY compensation>Inner set to 0.119mm.

If I add that to my settings I get the following
[image]

Oh Bingo.  Not quite the setting that jhlchu guessed at but close.  If you read the tooltip for that setting it should explain what is happening
[image]

drveni, 2022-10-01:
oh, you are right, then my deep apologies. I miss this. please close the topic. thx!