supermerill/SuperSlicer #3293: SuperSlicer_2.4.58.5 UI layout overrides permanently SuperSlicer_2.5.59.0 layout

NiiXXiiN, 2022-09-29:
### What happened?

I use both versions of SuperSlicer. Version 2.4.58.5 for work and 2.5.59.0 for tests. When i updated to 2.4.58.5 from 2.4.58.4 the new version overrode permanently the Arachne  options in print settings and made them invisible. The only way to make them reappear  is to erase the ui_layout folder in ..\AppData\Roaming\SuperSlicer

### Project file & How to reproduce

Use SuperSlicer_2.5.59.0 close the program, use SuperSlicer_2.4.58.5 and close the program. Then use again SuperSlicer_2.5.59.0.

### Version

SuperSlicer_2.4.58.5, SuperSlicer_2.5.59.0

### Operating system

windows 11

### Printer model

v-core 3

ruredi87, 2022-09-29:
yep i have same issue

GKSN, 2022-10-01:
same problem, look at issue #3094 . If you change the version in the "/Resources/ui_layout/default/version.ini" it works too ( don't know if it is the best option ).

stratiegery01, 2023-01-16:
same issue here. Has anyone found a fix for this?

stratiegery01, 2023-01-17:
> Did you read issue 3094 ?

I missed one of the entries which worked: https://github.com/supermerill/SuperSlicer/issues/3094#issuecomment-1262778729
Thanks!

NiiXXiiN, 2023-01-20:
> 
Yep, that's the way i got it working for now.