supermerill/SuperSlicer #3294: Contiguous Seam Massive Discontinuity

CCS86, 2022-09-29:
### What happened?

Am I doing something wrong here?

[image]

### Project file & How to reproduce

[Contiguous.3mf.txt](https://github.com/supermerill/SuperSlicer/files/9679389/Contiguous.3mf.txt)

### Version

2.4.58.5

### Operating system

Windows 10 x64

### Printer model

UMO+

jhlchu, 2022-09-30:
Seam algorithm is still being improved. When PrusaSlicer 2.5 changes get merged, the better seam algorithm will be added. Current workarounds are seam painting or seam sphere. See change log of PS 2.5 for details of improved seam algorithm.

jhlchu, 2022-09-30:
[image]
Left to Right: Seam Painting, Seam Blocker Painting, Seam Sphere, Untouched

neophyl, 2022-09-30:
The PS2.5 seam changes are not very good in many respects.  Yes they try and give a single line, the problem is even when you set aligned or rear and used to get a nice straight seam line it now does it curvy which is not desirable for many prints.  I suspect the prusa team will be altering it in 2.6 to allow the previous use of a proper straight seam.
So Im sort of hoping that the 2.5 changes arent ported in untouched.  Besides the seam sphere normally works so well if used.

CCS86, 2022-09-30:
> Left to Right: Seam Painting, Seam Blocker Painting, Seam Sphere, Untouched

If you look at my original project, I am using a seam sphere on each object to position the contiguous seam. This usually works for me, but in this case is failing badly.

My goal was to get a seam like this on every part:

[image]

neophyl, 2022-09-30:
with 2.5.59 with your project as it I get
[image]

Isnt that what you want ?

neophyl, 2022-09-30:
Ah I think that the seam sphere not doing anything is a previously posted bug, where the project isnt saving it right.  If I delete your seam sphere and add a new one it then works.
[image]

But if I save it and then re-open it it goes back to not doing anything.

CCS86, 2022-09-30:
Interesting, I'll give that a try!

neophyl, 2022-09-30:
Found the issue I remembered https://github.com/supermerill/SuperSlicer/issues/3153  funny as it was opened by jhlchu :)

CCS86, 2022-09-30:
This is odd. It still refuses for me.

I deleted all but one part model, and all seam spheres. then I created a new seam sphere, and re-added the other instances. 

It still refuses to put the seam where I want it and still puts the seam somewhere totally different for a small section:

[image]

neophyl, 2022-09-30:
I cant get it not to work in 2.5.59.  What's the red  and purple thing in the corner ?  the thing that also looks like its interrupting your skirt loop.

I've also noticed that combining painted areas of seam enforcers and blockers reacts in strange ways with the sphere.  At least until I figure out what its doing.  I  don't thing I've ever really used the paint on ones as they just weren't needed on anything I print.

CCS86, 2022-09-30:
> I cant get it not to work in 2.5.59. What's the red and purple thing in the corner ? the thing that also looks like its interrupting your skirt loop.
>
> I've also noticed that combining painted areas [...]

Sorry, that was me "superimposing" my seam sphere from the 3d view.

I'll try it in 2.5

CCS86, 2022-09-30:
Yup, 2.5.59.0 works perfectly. The issue lies in 2.4.58.5

[image]

julianjc84, 2022-12-22:
Interesting object.
it's hollow and has pretty much a full perimeter loops. So that means pretty much all the seams can be 100% hidden inside the object.
SS v2.59.9
seam aligned is my go to auto seam, and I touch it up rarely with seam painting. Not required for this object.
sometimes it can be hard to see where its putting the seams. With Feature type External Perimeters and Seams enabled you can see the seams very clearly.

https://user-images.githubusercontent.com/14962510/209240140-f6e6abbf-e805-45ee-885d-768325d3f094.mp4