supermerill/SuperSlicer #3295: Interior Infill & Solid Infill Overlap Request/Issue

kpdcompany, 2022-09-29:
**Bug description**
Interior Bridge Infill Needs Threshold and "OFF" switch
For the Solid Layer under the Top Solid infill, does NOT follow "solid_infill_overlap"

!!>> Project File <<!!
[Flow calibration- HTN.zip](https://github.com/supermerill/SuperSlicer/files/9679398/Flow.calibration-.HTN.zip)

**To Reproduce**
See Project File and Images

**Screenshots**

INTERIOR BRIDGE INFILL IS NOT NEEDED ON 50%+ INFILL
[image]

LAYER ABOVE BRIDGE INFILL IS CORRECT WITH solid_infill_overlap setting at 95"%, NEXT LAYER DOES NOT FOLLOW
[image]

(NEXT LAYER) LAYER BEFORE SOLID TOP, As you can see, no gap between lines.
[image]

**Desktop (please complete the following information):**
WIN 10
Troodon

neophyl, 2022-09-30:
That light blue coloured layer that is classed as bridge infill IS the first layer of your top shell.  It is NOT the same thing as the supporting dense layer setting.
Its classed as bridge infill BECAUSE it is bridging the infill.  It will be that unless infill is 100%, and as the only infill type that can be set to 100% is rectilinear that's not going to happen.  You have 4 top layers configured, count them as you go down each layer, red topfill, 2 layers of normal infill in purple and the blue layer as bridge infill.  4 in total.

The gaps between the solid infill (purple) are down to your flow settings being set to 90% for above bridges (that first solid infill layer is over the bridge infill layer hence using the 90% flow you have set).  Your topfill layer also has gaps between as that is also configured to 90% but thats hard to see as you have ironing enabled.  Turn ironing off and you will see it easily.  So the only one without gaps is that single solid infill layer between your topfill and the one above the bride infill.  SuSi is doing exactly what you have it configured to do.

I'm not sure yet as to why the supporting dense layer option is greyed out in your project though.

kpdcompany, 2022-09-30:
That Purple Layer has NO gaps as it should.  The SOLID INFILL SPACING is not being used.  There is 95% Overlap and it only works on a solid layer, not a top solid layer?  That makes zero sense.  Why would you want the solid layers to not be the same besides the top?  What I am also wanting is this bridge infill to be able to be disabled.  There is absolutely no reason why at 50% infill you cant just run a regular solid layer.  I have to die these extrusions down a LOT to get no over extrusion.  This is the only place it happens.  I do not understand why ONE solid layer is not being spaced out or effected by extrusion.  Layer 10 shows a sloid layer that can not be modified, this is the issue. this layer fills in so thick and can not be adjusted spacing wise like the other solid layers.  Even the Top Solid under the Ironing has a spacing gap according o this setting.  So why is it not applying it to ALL solid layers like it says?  If this was the first layer(in place of the bridge infill) this would make sense why its not spacing out but this is not needed or the case.  My point is, solid infill overlap is not for ALL solid layers like it says, I have solid layers configured to be a 95% overlap, and it is NOT doing this to all solid layers.  IF this is a logistical issue, then a setting for "Solid before top spacing" or something could be added.  Th amount of top layers should not matter in this example of my issue as it will do it before any top solids.  The bridge infill should have a setting for threshold or to turn it off, the fact there is this layer at all is quite odd unless printing very low infill.  I've only noticed this setting helping on Ironing calibration for low infill.

neophyl, 2022-09-30:
wall of text, please use punctuation and spacing as its difficult to follow.

Lets take this layer by layer.  First of all turn off the iron top layer temporarily so we can see what is going on.  Also lets just use a simple box for the example without any of the modifiers that you have.

Ok the top layer
[image]
You have top layer Flow set to 90%.  The spacing and width is set normally but you have configured Top fill flow to be 90%.  Can you see how that is resulting in the gaps between the lines ?

Ok next layer down
[image]
This is your second layer for the 4 configured for your top shell, its solid infill in rectilinear and it is touching as it does not use any of your modified flow settings.  This is equivalent of your layer 10.  Its the most 'normal' solid infill layer as its not being effected by any of the other settings you have configured for top or above bridges.

Next one the third 
[image]
This layer is also solid infill.  However the layer below that has been classed as bridge infill.  As such it is using the 90% flow rate you have configured for 'above the bridges'.  So it has gaps just like your top layer.  Follow so far ?  Everything is slicing just as you have it configured.

Next layer down, the last of your 4 configured top layers.  
[image]
As this is printed above infill this _automatically_ gets classed as bridge infill.  While it is true that it will probably print over the percentage infill you have selected just fine, it makes no difference.  Its still printing over gaps so its bridging, even if the gaps are small.  You may not agree with that logic, but its that way as the gap size could be anything depending on the pattern or infill chosen.  
The reason there are gaps between the lines is the setting you have selected for Bridge flow baseline of 'keep current flow'.  That is reducing the width of the lines to 0.32  The width is even less than that the top infill.  

Personally I don't really like the options around the bridge flow and I never use that keep current flow option as it never seems to give me a satisfactory result.  That's just me though, I'm sure some people use it fine.

I much prefer Prusa's simpler disabling of thick bridges.  The simulate prusa setting doesn't actually give the same results so I don't use that either.  That leaves us with normal width extrusions which result in double thick height too (the old thick bridge default), or use layer height which results in it trying to print very thin extrusions but at least the height is ok.  Prusa's just ignores the height but still gives a normal width.  At least in the gcode previewer.  It also gives acceptable results for printing.

Below that is your infill which is perfectly normal.  

Remember all this logic was in there BEFORE the option for supporting dense layer was added.  The slicer doesn't deliberately mess with you, it only seems that way sometimes.  Its best to understand exactly what is going on, what settings are currently in play at each layer or area and work it out logically from there.

That way you can target any feature requests or changes more specifically.  

It may be that the Solid fill Overlap setting is broken, it certainly seems to do nothing in 2.5.59.  Perhaps the merge with PS in this case.  That setting doesn't exist in my installs of 2.2.53 or 2.2.54 iirc so its a very new setting.  The top layer and above bridges flow have been there much longer.

I take it you have dialed in all the filament settings that could be effecting flow ?

kpdcompany, 2022-10-03:
Also I would say that my "supporting Dense Layer" setting that is greyed of, as in I cant select it to turn it on makes me think there is a decent bug in this setting.  Maybe this is why It is always on, you can no seem to turn it off or turn it on(default says its off but it is not) 

I should be able to check or uncheck this box.  This is not a feature request but a bug as originally stated. 

If I am wrong please tell me what setting I can change to be able to enable/disable this supporting dense layer.

neophyl, 2022-10-03:
You are still confusing things.

A supporting dense layer (when enabled) adds an EXTRA layer between the infill and the first of the top shells.  That supporting dense layer spacing is always 50%.  You noted that when the infill is 50% or above then you cant enable the supporting dense layer.  That makes sense as your infill % is already matching the supporting dense layer and so it doesn't do any good.

All the examples in this thread that you have posted do not show ANY use of supporting dense layer at all.  Stop calling it that as it is not supporting dense layer.

What you have been calling and confusing as supporting dense layer is the bridge infill layer.  As was explained above this is part of the top shell and is the bottom most layer of the top shell.  You **can not** at the moment turn that off (you have never been able to turn that off) so yes if you want the ability to turn that off then that needs to be a feature request.  There is NO setting to turn that off currently.  Its NOT a supporting layer, its part of your Top shell layers.

Yes depending on the settings it can be over extruded (this is the 'old' default and the equivalent of the prusa thick bridges behavior).  Or it can be under extruded if you keep the flow or if you set layer height you get a very densely packed layer *if you have your settings correct).  If you don't you tend to get gaps in that too.

As was also explained above the gaps in some of the top shell layers, namely the layer above the bridge infill and the top infill are caused by your 90% flow settings that over ride for those particular layers. Try following the example I posted above.

kpdcompany, 2022-10-03:
Also have zero Idea how to change my post to a request, the title already says as much.

neophyl, 2022-10-03:
It’s called bridge infill because it **bridges** over the infill below. 
It’s might be unintuitive to you but not to others. 
Edit the title of the issue, to make the title more specific to what you want. 
I don't have any further time to spend on this.  You have been given an explanation, if that doesn’t meet with your expectations then as I see it you have 3 choices

1- you hope that the developer agrees with you and will change the operation of the software to “fix” it. Hope patience is a strong suite. 

2- download the source code, modify it to behave like you want and compile a version that works like you want. 

3- find some slicer software that works like you want and start using that instead. 

Good luck.

kpdcompany, 2022-10-07:
> It’s called bridge infill because it **bridges** over the infill below. It’s might be unintuitive to you but not to others. Edit the title of the issue, to make the title more specific to what you [...]

Well let me give you some advice, how bout not being a rude dickhead...