supermerill/SuperSlicer #3297: No bottom layers over bridging perimeters

RealDeuce, 2022-09-30:
### What happened?

When all perimeters in a layer are bridges, no additional layers are printed on top of them, so the vertical thickness is only one layer.

[image]

### Project file & How to reproduce

[stream_for_skull.zip](https://github.com/supermerill/SuperSlicer/files/9679954/stream_for_skull.zip)

Slice, look at the centre "teardrop" on layer 43 of the large bottom piece.  Above this, onle the bridged section in the middle has solid infill, the perimeters themselves do *not* have solid infill over them.

This issue was also present in an older nigthly, but I upgraded to see if it was already fixed.

### Version

2.5.59.0 @ f8c9106c01

### Operating system

FreeBSD 12.3-RELEASEp5

### Printer model

Voron v2.4 with StealthBurner

jhlchu, 2022-09-30:
Not sure what outcome you are expecting. I'm guessing you want solid fill over the overhang (blue) perimeter on layer 43-44. You can achieve that be un-checking `Print Settings > Perimeters & Shell >Ensure Vertical Shell Thickness`.
[image]
[image]

RealDeuce, 2022-09-30:
I would not expect that "Ensure Vertical Shell Thickness" would cause vertical shells that are 0.2mm thick.  Those perimeters are on the bottom of the object, and there is only a single layer of material instead of the configured 8 (or 1.6mm).

I'm confused at why you would think that a single bottom layer in that position is expected.

RealDeuce, 2022-09-30:
It looks like an issue with solid_over_perimeters... if that's set to 0 (disabled) I get 7 layers of solid infill (which including the perimeters, gives me 8 layers of thickness), but now matter how high it is set, I never get any.  I would expect to get three, since that's what solid_over_perimeters is set to.

jhlchu, 2022-09-30:
Without `Ensure Vertical Shell Thickness` there are 8 bottom layers above the bridge. With `No Solid Infill Over = 0` also gives similar results.

RealDeuce, 2022-09-30:
Yes, disabling ensure_vertical_shell_thickness disables solid_over_perimeters (ie: sets it to zero), which is why I suspect a but in solid_over_perimeters (no matter how you disable the setting, the problem goes away) rather than an issue with ensure_vertical_shell_thickness since they issue does not occur when solid_over_perimeters is set to zero.

neophyl, 2022-09-30:
Solid over perimeters was added relatively recently and it was put in to prevent the solid infill being added on sloping perimeters.  Some models no matter if you had Ensure Vertical Shell thickness=OFF still put the solid infill in, when it wasnt needed or wanted.

The Solid perimeters value was for stopping it doing that, so zero in effect set the 'old' Solid Infill behaviour and anything above Zero uses whatever the algorithm Merill added to try and make it NOT add in the solid infill.  The larger the number the more if removes the solid infill.

RealDeuce, 2022-09-30:
The larger the number the *less* it removes the solid infill.  The number is how many layers there should be between the inside of the model (ie: normal infill) and the air on the outside of the model... which is to say there is the perimeter itself, then (solid_over_perimeters - 1) layers of infill on top of it.

When solid_over_perimeters >= bottom_solid_layers (or top for features on the top), it should have the same effect as setting solid_over_perimeters to zero.

rajhlinux, 2023-08-04:
> ### What happened?
> When all perimeters in a layer are bridges, no additional layers are printed on top of them, so the vertical thickness is only one layer.
> 
> [image] [ [image] ](https://user-images.githubusercontent.com/9758938/193170935-52fe8b8b-b3b2-4385-881f-384a40cce564.gif) [ ](https://user-images.githubusercontent.com/9758938/193170935-52fe8b8b-b3b2-4385-881f-384a40cce564.gif)
> 
> ### Project file & How to reproduce
> [stream_for_skull.zip](https://github.com/supermerill/SuperSlicer/files/9679954/stream_for_skull.zip)
> 
> Slice, look at the centre "teardrop" on layer 43 of the large bottom piece. Above this, onle the bridged section in the middle has solid infill, the perimeters themselves do _not_ have solid infill over them.
> 
> This issue was also present in an older nigthly, but I upgraded to see if it was already fixed.
> 
> ### Version
> 2.5.59.0 @ [f8c9106](https://github.com/supermerill/SuperSlicer/commit/f8c9106c0119b6373b8ace5436a756a0c0003dd8)
> 
> ### Operating system
> FreeBSD 12.3-RELEASEp5
> 
> ### Printer model
> Voron v2.4 with StealthBurner

How did you install on FreeBSD?

RealDeuce, 2023-08-08:
I just pushed my repo to here: https://github.com/RealDeuce/SuperSlicer

RealDeuce, 2023-08-08:
> How did you install on FreeBSD?

Just force-pushed to https://github.com/RealDeuce/SuperSlicer so hopefully you didn't already have it checked out.