supermerill/SuperSlicer #3298: SuperSlicer not running on Intel MacOs Ventura 13.0

iamshaunicus, 2022-09-30:
### What happened?

Superslicer crashes on startup. Below is the crash log

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [658]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-09-30 12:02:51.5086 +0100
OS Version:            macOS 13.0 (22A5295i)
Report Version:        12
Bridge OS Version:     7.0 (20P50329e)
Anonymous UUID:        E23F89C7-D5E5-4D13-5F1D-FEA35D1644F1

Time Awake Since Boot: 30 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: <EA3F7209-F390-3268-9C90-D8081DCEDC30> /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	    0x7ffb2662753a __abort_with_payload + 10
1   dyld                          	    0x7ffb26640687 abort_with_payload_wrapper_internal + 82
2   dyld                          	    0x7ffb266406b9 abort_with_payload + 9
3   dyld                          	    0x7ffb265c980c dyld4::halt(char const*) + 375
4   dyld                          	    0x7ffb265c6596 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4502
5   dyld                          	    0x7ffb265c51bd start + 2125

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b11c4558  rdx: 0x00007ff7b11c49c0
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7b11c45a0  rsp: 0x00007ff7b11c4558
   r8: 0x00007ff7b11c45c0   r9: 0x0000000000000000  r10: 0x0000000000000075  r11: 0x0000000000000246
  r12: 0x0000000000000075  r13: 0x00007ff7b11c49c0  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x00007ffb2662753a  rfl: 0x0000000000000246  cr2: 0x0000000111fbf000
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
    0x7ffb265bf000 -     0x7ffb266519d7 dyld (*) <db000c0b-1de2-33d3-a29e-b22dbc068d3a> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 0
[... 131 lines cut ...]
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

### Project file & How to reproduce

Usual unidentified dev warning.
Sudo xattr -cr /Applications/SuperSlicer.app in terminal to resolve.
Upon opening SuperSlicer crashes and produces the above report.

### Version

late 2.4 edition

### Operating system

macOS 13

### Printer model

X5SA / Klipper

Laminador, 2022-09-30:
Intel Mac version don´t work on Mojave 10.14.6 either even  with xattr -cr /Applications/SuperSlicer.app permission.

abjorck, 2022-10-09:
`brew install zstd` made it run for me

Henk72, 2023-01-01:
> > `brew install zstd` made it run for me

Thank you, this allowed me to run SuperSlicer on Mac OS 10.14.6 Mojave!