supermerill/SuperSlicer #3303: ConfigBase::get_abs_value(): nozzle_diameter need to has the extuder id to get the right value, but it's not available

Jackymancs4, 2022-09-30:
### What happened?

The original STL sliced just fine.
After I applying a multimaterial painting modifier to the object, and cliking on Slice Now, two message box appear with the message:
`ConfigBase::get_abs_value(): nozzle_diameter need to has the extuder id to get the right value, but it's not available`

Screenshot for reference:
[image]

I opened the very same 3mf file on PrusaSlicer 2.5 and it worked.

Thank you.

### Project file & How to reproduce

1. Open the project
[Goose_full_multicolor.zip](https://github.com/supermerill/SuperSlicer/files/9687986/Goose_full_multicolor.zip)
2. Click on Slice Now
3. Close the errors.

### Version

Version 2.4.58.5

### Operating system

Windows 10

### Printer model

Geeetech A20M

whi-tw, 2022-10-04:
I'd just like to add another datapoint to this: I've got the exact same result

### Project file & How to reproduce
1. Open the project: [angryusb.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9706434/angryusb.3mf.zip)

1. Click on Slice Now
1. Close the errors.

### Version
Version 2.4.58.5

### Operating system
MacOS 12.6 (M1 Pro)

### Printer model
Voron 2.4 R2 (ERCF)

meganssmith, 2022-10-08:
another data point.  Paint a model so it is 2 color and then slice it.

ChristianReil, 2022-10-09:
Is there any solution? I have the same issue...

ChristianReil, 2022-10-09:
We found a workaround for that issue: 

Choose the Perimeter Parameter in mm and not in % and then it works :-)

Dreistein75, 2022-10-18:
@ChristianReil can you please tell me more explicitly what Perimeter Parameter you changed to work around this problem?

Dreistein75, 2022-10-18:
New data point: The same error does not happen in the beta version of SuperSlicer 2.5. But in SuperSlicer 2.5 the slicer just ignores all multi-colour paintings (with smart fill)....

ChristianReil, 2022-10-18:
> @ChristianReil can you please tell me more explicitly what Perimeter Parameter you changed to work around this problem?

In the print settings tab -> Width & Flow -> perimeter:  write in a mm value (example 0,48mm) and no % Value (Example 120%) then it will work. 

So I think there is the error...

Dreistein75, 2022-10-23:
> > @ChristianReil can you please tell me more explicitly what Perimeter Parameter you changed to work around this problem?
> 
> In the print settings tab -> Width & Flow -> perimeter: write in a mm value (example 0,48mm) and no % Value (Example 120%) then it will work.
> 
> So I think there is the error...

Thank you so much!! Worked very well, but it's a bit tedious to calculate all extrusion widths...

Imburr, 2023-03-15:
Still an issue in latest version.

mochoandre, 2023-04-10:
Having same issue

Maker-M, 2023-04-19:
Same problem for me:

 - Superslicer 2.4.58.5
 - Windows 10
 - Tenlog TLD3 Idex printer

I would like to add an information:

 - Slice impossible when the perimeter parameter is in % 
 - I change the parameter to mm instead of %, the slice becomes possible
 - I change again the parameter in % instead of mm, the slice is now possible in %.
 - I close Superslicer, I restart it, the slice in % causes the original error again, and I have to start the manipulation again.

It would really help a lot to correct this problem.

matthewj301, 2023-04-23:
Same issue

ChristianReil, 2024-02-10:
> > @ChristianReil can you please tell me more explicitly what Perimeter Parameter you changed to work around this problem?
> 
> In the print settings tab -> Width & Flow -> perimeter: write in a mm value (example 0,48mm) and no % Value (Example 120%) then it will work.
> 
> So I think there is the error...

@supermerill  Can you fix this for the % line width?

supermerill, 2024-02-11:
> @supermerill Can you fix this for the % line width?

it's still broken?
In the nightly (latest release has some problem loading width & spacing), I can't see any issues at first glance.

supermerill, 2024-02-11:
ah, it's still here

closed 2024-05-03 by supermerill

dancollins, 2025-04-25:
I'm running 2.5.59.13, and swapping from relative to absolute width in print settings fixed this. I should note this seems related to painting - as multi-part MMU prints didn't trigger this for me. I am assuming, since this was completed in Feb 24, that my July 24 release includes the fix!

Can someone re-open this?

supermerill, 2025-05-26:
> I'm running 2.5.59.13, and swapping from relative to absolute width in print settings fixed this. I should note this seems related to painting - as multi-part MMU prints didn't trigger this for me. [...]

It's fixed in 2.7.61, not 2.5.59.13
do you have a project to share where the bug is still here?