supermerill/SuperSlicer #3304: non-planar support extrusions which the nozzle crashes into

doctor-bob, 2022-10-01:
### What happened?

i noticed the z axis moving while laying down support. then i heard the nozzle running into the support. the slicer shows these extrusions.

what is going on here? i might have heard that sound before, but i explained this with overextruded layers sticking out...

### Project file & How to reproduce

[image]

### Version

Version 2.5.59

### Printer model

ender3 with BTT skr v3

doctor-bob, 2022-10-01:
i am new to this github stuff, so if you want me to post further information, please tell me what to do! :)

neophyl, 2022-10-02:
In future can you please just attach the project file from SuSi.  File>Save as.  That creates a 3mf file whoich already contains the model and ALL the settings in use as well as as modifiers, variable layer heights, placements etc.

We can then just open it and slice with the same settings you are using.  Theres always the danger that when we add things manually from the files people provide that something will effect trying to recreate an issue.

In this case though its very simple.  You have sawtooth pattern selected for the support material interface (Print Settings>Support Material>Options for support material interface>Pattern).

doctor-bob, 2022-10-02:
I will look into the settings, thank you.

as much as I like the idea of 3d extrusion paths, what made me look at this as a bug is the fact that the nozzle crashes into those bumps constantly.
I had similar problems before when experimenting with layer and support heights.