supermerill/SuperSlicer #3306: Add lightning infill from PrusaSlicer?

mlewis-everley, 2022-10-02:
In PrusaSlicer 2.5, lightning infill has been added, which is great for printing items that don't really require infill. I am using SuperSlicer 2.5.59 but lightning infill does not appear to be available?

It would be great to be able to use lightning infill in SuperSlicer the same way I use it in PrusaSlicer (just select it as an infill type from the dropdown).

I have checked the existing issues, discussions and pull requests and cannot find any other requests for this feature, so I don't believe it has been raised before?

user897943, 2022-10-13:
Absolutely. Also bring over Arachne which does perimeters so much better.

mlewis-everley, 2022-10-13:
Yep, Arachne is already there, but it is disabled by default, @user897943 you have to:

1. Open Print Settings
2. Select "Perimeters & Shell"
3. Navigate to the bottom of the screen ("Arachne perimeter generator") and switch from "Classic" to "Arachne"

user897943, 2022-10-15:
Thank you but how is it possible to update SuperSlicer to that version on chocolatey? choco upgrade SuperSlicerChocolatey v1.1.0Upgrading the following packages:SuperSlicerBy upgrading, you accept licenses for the packages.superslicer v2.4.58.5 is the latest version available based on your source(s). Chocolatey upgraded 0/1 packages. See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log). It’s not there yet!? Sent from Mail for Windows From: Morven Lewis-EverleySent: Thursday, October 13, 2022 1:54 PMTo: supermerill/SuperSlicerCc: user897943; MentionSubject: Re: [supermerill/SuperSlicer] Add lightning infill from PrusaSlicer? (Issue #3306) Yep, Arachne is already there, but it is disabled by default, @user897943 you have to:Open Print SettingsSelect "Perimeters & Shell"Navigate to the bottom of the screen ("Arachne perimeter generator") and switch from "Classic" to "Arachne"

user897943, 2022-10-15:
Ok so I went to github and extracted the ZIP, launched the app, on the title bar it says:
 SuperSlicer 2.5.59 based on PrusaSlicer... so this should have Arachne on the perimeters, I am using Expert mode, looked everywhere (including at the bottom) and there is nothing about Arachne, no dropdown menu to change from classic to Arachne. What am I missing!?

yellowsink, 2022-10-15:
I love lightning infill in Cura, +1 for this, currently the main sticking point for me switching to this

mlewis-everley, 2022-10-15:
@yellowsink It is not the end of the world, I have found "Dense Infill Layer" works pretty well for most use cases (it basically does automatically what I was doing manually prior to lightning).

I have found though that dense infill layer does sometimes use more filament or still result in extrusion issues on top layers compared to lightning

HinataKato, 2023-01-23:
It's in nightly build now, but look kana weird
[image]

mlewis-everley, 2023-01-25:
> It's in nightly build now, but look kana weird [image]

Yep, I have just downloaded the latest nightly build. If you are unaware how to get these, you can download them from the "actions" tab: https://github.com/supermerill/SuperSlicer/actions

Please be aware though, these are development builds, so may be less stable than a fully tested "release".

mlewis-everley, 2023-01-25:
Opps, closed by accident.