supermerill/SuperSlicer #3307: Auto-Arrange add by height

DrewCipher505, 2022-10-02:
Adding "By Height" auto arrangement may help to reduce travel moves and by extension, print times.

Addition of an option in the right click menu of auto arrange to select "height" as a parameter to group objects with similar height in a gradient towards possibly even a selection of corners

OldGuyMeltsPlastic, 2022-11-22:
I had this exact same thought today looking at the randomly arranged full plate of V01 parts that I just finished printing!

If the tallest objects and/or object parts/corners/etc are clustered together, maybe near the center of the bed, then travel moves can be reduced.

3DAixperts, 2023-02-04:
That´s exactly what I´m looking for right now. Would be a very nice feature to optimize the prints.
@DrewCipher505 can you add a [FR] to the name of this issue?