supermerill/SuperSlicer #3308: wipe_tower_no_sparse_layers prints in a midair

rednaxal, 2022-10-02:
### What happened?

When the wipe_tower_no_sparse_layers setting is enabled for some reason it prints the color change layer in a midair above the wipetower.

[image]

### Project file & How to reproduce

Project file: [Bug.zip](https://github.com/supermerill/SuperSlicer/files/9692916/Bug.zip)

### Version

2.4.58.5

### Operating system

Windows 10

### Printer model

Kingroon KP3S

joey2point0, 2022-10-27:
Wondering if this is fixed in PS release 2.5. I know the feature is still functional in SS release 2.3.57.

spikeygg, 2022-11-11:
This issue is a duplicate of [2964](https://github.com/supermerill/SuperSlicer/issues/2964)

It was not fixed by 2.5.59.