supermerill/SuperSlicer #3312: Pause time showing sminutes remaining as seconds M117

kentishduffer, 2022-10-03:
2.4.58.5

Windows 11
Creality Ender3 v2

When using the Pause function (M600) with M117 enabled the 'Pause in hms' shows seconds instead of minutes.

Create a colour change with M600.
In Printer Settings select 'Print Remaining Times' and Method M73 & M117
Slice object and save GCode.
Print object.
Display reads seconds instead of minutes.

e.g. Display should read 0h13m0s
       Display is showing 0h0m13s

See attached screenshot of GCode
[image]

kentishduffer, 2022-11-24:
Can anyone help with this?

luismanson, 2022-12-03:
I've found the same issue, however I did a filament change some prints ago and the time is still shown as reaming time. 
@kentishduffer  did you try Prusa slicer?

kentishduffer, 2022-12-03:
@luismanson I haven’t tried Prusa. The remaining time is correct, it’s just the pause time which is incorrect.

Rneil-ca, 2023-01-21:
I am see this issue as well.  The first pause in time is correct, but after that it shifts:

M73 P0 R547
M73 C382
M117 Pause in 6h22m25s / 9h7m7s
;TYPE:Custom
; custom gcode: start_gcode
START_PRINT BED=70 EXT=215
; custom gcode end: start_gcode
;_TOOLCHANGE 0
M107 ; disable fan
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
; custom gcode: before_layer_gcode
;BEFORE_LAYER_CHANGE
;0.2
G92 E0
; custom gcode end: before_layer_gcode

G1 E-0.2 F2100 ; retract
M73 P0 R546
M73 C6
M117 Pause in 0h6m22s / 9h6m47s

kentishduffer, 2024-03-27:
@supermerill Could you have a look at this issue please? Thanks