supermerill/SuperSlicer #3313: Pressure equalizer from PrusaSlicer

LordKillZone, 2022-10-03:
**Is your feature request related to a problem? Please describe.**
Are there any plans to add pressure equalizer from Prusa Slicer into SuperSlicer.

**Describe the solution you'd like**
Adding Pressure Equalizer

**Describe how it would work**
Its equalizing pressure in nozzle, adding gradient speed change.

**Additional context**
I would like to test this feature, maybe it will make my prints a lot prettier.

tylerinpdx, 2022-10-16:
Just use the linear advance, and deal with a ridiculous amount of noise If you have a long bowden system it's pretty much the only way besides shortening it and wrapping a shrink wrap to so it doesn't stretch as much

MrBlackmidi, 2022-11-08:
The advantage with Pressure Equalizer is that it isnt required to mess with firmware

meltiseugen, 2022-11-22:
I'm also interested to know when this will be implemented in SS

MrBlackmidi, 2022-11-23:
Also I think pressure equaliser may be useful for really flexible materals