supermerill/SuperSlicer #3314: Multicolor Prints produce negative values for certain G1 F commands when printing wipe tower

HitLuca, 2022-10-03:
### What happened?

I was trying to print a multicolor calibration cube, with M600 filament changes at specified layer heights, but my print failed. Debugging the issue led me to find that right before the M600 command, the slicer produced a bunch of `G1 F-2147483648` entries, which to me look like integer overflows.

my gcode file is here https://www.zerobin.tk/?166135e858db7cc6#2Lp3Y71PBkizREedCPnnhmrUC8sYCKGhK4wkoixj6j5w and I can reproduce it consistently with new multicolor prints.

The affected section is below 
```raw
;--------------------
; CP TOOLCHANGE START
; toolchange #1
; material : PLA -> PLA
;--------------------
M220 S100
; CP TOOLCHANGE UNLOAD
;WIDTH:0.9
G1  X65.552 Y143.489  
G1  X98.227  E2.7942
SET_PRESSURE_ADVANCE ADVANCE=0
G1 F1580
G1  X104.809  E0.686
G1 F1628
G1  X111.591  E0.7068
G1 F1723
G1  X118.771  E0.7484
G1 F1819
G1  X123.302  E0.4723
G1  Y144.389  F7200
G1 F1819
G1  X120.254  E0.3177
G1 F1891
G1  X112.376  E0.8211
G1 F1963
G1  X104.198  E0.8523
G1 F2082
G1  X95.522  E0.9043
G1 F2250
G1  X86.148  E0.977
G1 F2369
G1  X76.275  E1.029
G1 F2393
G1  X66.302  E1.0394
;WIDTH:0.75
M104 S215
G1  Y145.289 
G1 F3000
G1 F-2147483648
G1 F3000
G1 F-2147483648
G1 F3000
G1 F-2147483648
G1 F3000
G1 F-2147483648
G1 F2000
G1  Y145.214  F2400
G4 S0
; custom gcode: end_filament_gcode
; Filament-specific end gcode
; custom gcode end: end_filament_gcode

; custom gcode: toolchange_gcode
;Change Temperature to the next required one 
M104 S215
M117 change to 1 SUNLU Silk PLA+ Rainbow
M600
; custom gcode end: toolchange_gcode
```

If needed, I can add my printer configuration file since I don't know exactly what is embedded in the Project file I attached below

### Project file & How to reproduce

[Calibration cube.zip](https://github.com/supermerill/SuperSlicer/files/9698782/Calibration.cube.zip)

### Version

2.4.58.5

### Operating system

Window s10

### Printer model

Eryone Thinker SE

HitLuca, 2023-04-22:
Possible workaround found and added as comment on the linked Prusa Slicer issue

Plumpkatt1, 2023-07-08:
> ### What happened?
> I was trying to print a multicolor calibration cube, with M600 filament changes at specified layer heights, but my print failed. Debugging the issue led me to find that right before the M600 command, the slicer produced a bunch of `G1 F-2147483648` entries, which to me look like integer overflows.
> 
> my gcode file is here https://www.zerobin.tk/?166135e858db7cc6#2Lp3Y71PBkizREedCPnnhmrUC8sYCKGhK4wkoixj6j5w and I can reproduce it consistently with new multicolor prints.
> 
> The affected section is below
> 
> ```
> ;--------------------
> ; CP TOOLCHANGE START
> ; toolchange #1
> ; material : PLA -> PLA
> ;--------------------
> M220 S100
> ; CP TOOLCHANGE UNLOAD
> ;WIDTH:0.9
> G1  X65.552 Y143.489  
> G1  X98.227  E2.7942
> SET_PRESSURE_ADVANCE ADVANCE=0
> G1 F1580
> G1  X104.809  E0.686
> G1 F1628
> G1  X111.591  E0.7068
> G1 F1723
> G1  X118.771  E0.7484
> G1 F1819
> G1  X123.302  E0.4723
> G1  Y144.389  F7200
> G1 F1819
> G1  X120.254  E0.3177
> G1 F1891
> G1  X112.376  E0.8211
> G1 F1963
> G1  X104.198  E0.8523
> G1 F2082
> G1  X95.522  E0.9043
> G1 F2250
> G1  X86.148  E0.977
> G1 F2369
> G1  X76.275  E1.029
> G1 F2393
> G1  X66.302  E1.0394
> ;WIDTH:0.75
> M104 S215
> G1  Y145.289 
> G1 F3000
> G1 F-2147483648
> G1 F3000
> G1 F-2147483648
> G1 F3000
> G1 F-2147483648
> G1 F3000
> G1 F-2147483648
> G1 F2000
> G1  Y145.214  F2400
> G4 S0
> ; custom gcode: end_filament_gcode
> ; Filament-specific end gcode
> ; custom gcode end: end_filament_gcode
> 
> ; custom gcode: toolchange_gcode
> ;Change Temperature to the next required one 
> M104 S215
> M117 change to 1 SUNLU Silk PLA+ Rainbow
> M600
> ; custom gcode end: toolchange_gcode
> ```
> 
> If needed, I can add my printer configuration file since I don't know exactly what is embedded in the Project file I attached below
> 
> ### Project file & How to reproduce
> [Calibration cube.zip](https://github.com/supermerill/SuperSlicer/files/9698782/Calibration.cube.zip)
> 
> ### Version
> 2.4.58.5
> 
> ### Operating system
> Window s10
> 
> ### Printer model
> Eryone Thinker SE

I am having the very exact same issue when using color changing commands such as t1 t0 t3 and so on. My print fails just before the t command and gives this exact error and my g code is showing the exact same negative values. I will try to test this workaround to see if it successfully allows me to do a color swap. This is very frustrating as I have tested various versions of prusa as well as super slicer.