supermerill/SuperSlicer #3320: Ortholinear camera

CarnageMarkus, 2022-10-05:
Currently, the camera supports only perspective projection.
When slicing object into halves, it's hard to judge because of perspective projection "deforms".
Most 3D software I used has option to switch between perspective/orthogonal projection.

**What to do**
Add extra options under View -> Orthogonal projection / Perspective projection

**Describe how it would work**
Technically, this should be quite small change, as we only need to swap active projection matrices for the renderer. And/Or possibly remember last user choice.

Thank you for consideration.

CarnageMarkus, 2022-10-05:
Thanks, didn't know it was hidden there. That's 4 clicks away. Could we still get a shortcut for that in "View" toolbar ?

neophyl, 2022-10-06:
Its there as thats where it is in PS too.  Been there forever.  IIRC theres been requests over on the PS github for some easier way of switching between too.

plampix, 2022-10-19:
Just hit 'k' to switch.