supermerill/SuperSlicer #3324: FEATURE: Add a Non-Solid Support Interface Patern

inspectionsbybob, 2022-10-06:
### Version
2.4.48.5

### Operating system type + version
NA

### Behavior
I want to Minimize the Surface that touches my Models.  Currently, the only options are SOLID SURFACES which bond really well to the model which is unwanted.  Cura offers Zigzag, lines, grids, Triangles, and Concentric. at a LESS THAN SOLID value.

Solid works great for PVA (Soluble) Where the PVA gives the PLA a nice surface, PLA PLA is not good, even with .2+mm Z gap as to get good layer adhesion, running HOT means good bonding, even where you don't want it.

(Any other options to help make supports easier to remove that DOESN'T HURT LAYER ADHESION?)

_Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

neophyl, 2022-10-06:
No solid surfaces is not the only option.  Have you tried adjusting the support interface spacing ?  As well as adjusting the support pattern spacing the interface can also be adjusted.  You can set it to quite wide widths.   SuSi also has several patterns like concentric and sawtooth for that last actually touching layer.  Sawtooth puts little spikes upwards so has minimal contact with the object above.  Personally I don't like it but it might suit you.

For example default with a 2mm interface spacing 
[image]

inspectionsbybob, 2022-10-11:
Will look at that tomorrow.
My big problem is on large surfaces like bowls... what I had been using was
sticking too well and taking the bottom/outside surface of the bowl off
when I tried to pry the interface layer(s) off.

Tried lowering my temperature, and had adheasion problems.  Nozzle clipped
something as it was cooling and shifted a hunk...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016