supermerill/SuperSlicer #3327: "THe Wipe Tower is only supported for multiple objects if they are sliced equally.

ssjbardock, 2022-10-08:
### What happened?

I added an extra benchy to a full plate of multi-material prints, to get some use out of the wipe material (wipe into print), and I'm getting this error that won't let me slice.
[image]
[image]

Here is the plate in question.
[squirtle_multi_arms.zip](https://github.com/supermerill/SuperSlicer/files/9740199/squirtle_multi_arms.zip)

### Project file & How to reproduce

[squirtle_multi_arms.zip](https://github.com/supermerill/SuperSlicer/files/9740201/squirtle_multi_arms.zip)

Seems to occur when any 'non multi material split' models are added to a plate with 'multi-part' models.

### Version

2.5.59

### Operating system

Windows 10

wjsteele, 2024-03-03:
This is because the extruders have different settings.  For example:

[image]

tbrummel, 2025-01-07:
> This is because the extruders have different settings. For example: [...]

These are just limits for the nozzle.  The actual print layer height is set in print settings and is the same across the nozzles so I submit that these settings should not be triggering the error.  I know that they are -- you are correct.  However, they shouldn't as they don't determine the sliced layer heights.  I'm playing with multi-extruder prints on the XL with the latest (Dec 2024) Slicer and ran into this.