supermerill/SuperSlicer #3333: Infill failed

archerne, 2022-10-10:
### Version
2.5.59.0

### Operating system type + version
Ubuntu 22.10

### Behavior
When trying to slice https://www.printables.com/model/270984-wall-hook I get an error of "infill failed"

#### Project File (.3MF) where problem occurs
Project file can be found at the link above

tomohulk, 2022-11-02:
I just came across this same issue today as well, for me it was with the support auto generation.  I added 1 degree to the overhang threshold, and fixed it.

dunk7, 2022-11-06:
> I just came across this same issue today as well, for me it was with the support auto generation. I added 1 degree to the overhang threshold, and fixed it.

That fixed it for me too

aristoblis, 2025-02-03:
I ran into this problem using the Anycubic Slicer Next.  Turning off the Beam Interlocking feature fixed it for me.

jdg511, 2025-04-01:
I have this problem on the latest orca slicer

supermerill, 2025-04-02:
I don't have any project file, so I can't test.
If it's the support, it's duplicate of #4568
If it's not the support (or the latest nightly don't solve it), please give me your project file.