supermerill/SuperSlicer #3334: MacOS doesn't work

mcdfam1, 2022-10-11:
### What happened?

I can't get either version to install on MacOS 12.3.

the "_arm" version won't install, and the other crashes on open, even after "xattr -cr /Applications/SuperSlicer.app"

### Project file & How to reproduce

installer from site, SuperSlicer_2.4.58.5_macos_220924.dmg

### Version

2.4.58.5

### Operating system

macOS 12.3

eifeldj, 2022-10-12:
On my intel mac with macos 13.0 I cant open SuSl 2.4.58.5. 
2.3.57.9 is running good.

jdknoll, 2022-10-14:
I ran into this issue. I was able to get around it by doing a `brew install zstd`. Installation through homebrew is still needed even though it's not in the release notes anymore. Unfortunately, you need to install zstd through homebrew as installing through MacPorts doesn't seem to help.

fixing-it, 2022-10-22:
@jdknoll this helped, thank you!

Rhastlyn, 2022-10-30:
I'm experiencing this as well on a MacBook Pro with the Apple M1 chip running MacOS 13.0, it was working well with SuperSlicer v2.3.57.11.  After installing 2.4.58.5 and allowing the app to run through Privacy and Security, it immediately crashes with the error message. "SuperSlicer Quit Expectedly"