supermerill/SuperSlicer #3337: Invalid value provided for parameter top_infill_extrusion_width:

Stooovie, 2022-10-11:
### What happened?

SuperSlicer is generating gcode that fails to print, and the Gcode Viewer says it has "Invalid value provided for parameter top_infill_extrusion_width:".

[Here's the generated .gcode file](https://0x0.st/ovsp.12mm_210). It's the built-in 0.12mm preset for Ender 3, and default Ender 3 printer setting. 

### Project file & How to reproduce

[Clit Keychain.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9759811/Clit.Keychain.3mf.zip)

### Version

2.5.59

### Operating system

MacOS 12.6

### Printer model

Ender 3

neophyl, 2022-10-12:
When I load your gcode into the 2.5.59 gcode viewer on my win 10 machine it opens up fine and doesn't display any errors.

Similarly your project loads and slices fine and the gcode output from that is also ok.  I would note that you should really turn on snug supports and the outer parts of your model aren't supported and start printing in mid air so I would expect some loose filament stuck to the print and possibly an outright fail.

When loaded the printer shows as a modified Ender-3 and not a default stock Ender-3.  General, Custom Gcode, Machine Limits and Extruder all have changes marked from the default.  For example its using Klipper firmware.  Now I don't know if that's what you have configured but it is certainly not a 'default' Ender profile.

Stooovie, 2022-10-12:
You're right about the modifications. Yet the gcode fails to open with the Gcode viewer on my end for some reason. And it creates the same error even with 100% default profiles (mea culpa on sending the gcode created with non-100% default profiles, sorry).