supermerill/SuperSlicer #3338: With klipper as host type it is not possible to add API key

crstian19, 2022-10-12:
### What happened?

When you set authentication in fluidd, in order to upload gcode you have to add an API Key to superslicer but if you select Klipper as host type the API Key field does not appear.

[image]

Selecting Octoprint instead of Klipper works but I think it is not the easiest way.

### Project file & How to reproduce

1. Enter Physical Printer
2. Try to add API KEY

### Version

2.4.58.5

### Operating system

Arch Linux 5.19.13-arch1-1 x86_64

### Printer model

Ender 3

Fisheiyy, 2022-10-13:
You dont need an api key

crstian19, 2022-10-13:
> You dont need an api key

If you enable authentication, without api key superslicer cannot connect to klipper and returns 401 error.

Fisheiyy, 2022-10-13:
> > You dont need an api key
> 
> If you enable authentication, without api key superslicer cannot connect to klipper and returns 401 error.

You need to put [octoprint_compat] in your moonraker.conf file and restart moonraker and it will work, you don't need an api key

crstian19, 2022-10-13:
Again, please read the issue entirely. **Moonraker has authentication enabled, an API key field is needed.**

Fisheiyy, 2022-10-13:
> Again, please read the issue entirely. **Moonraker has authentication enabled, an API key field is needed.** The only way to do so right now is selecting Octoprint in Superslicer instead of Klipper

Maybe this is an isolated issue to you because I have no problem with it, I put [octoprint_compat] section in my moonraker.conf, restarted moonraker and it worked in superslicer with the klipper option, I did put the port after the ip, all I put was 192.168.1.69:7125 and it has been working fine

Fisheiyy, 2022-10-13:
[image]

CharlesScoville, 2026-01-03:
As of 2026 this is still an issue IMO.

@Fisheiyy. With respect, I feel that you don't appreciate what is happening with this. Just because there is a way to make things work as you have shown, does not mean it's supposed to work that way regardless. The API key is supposed to be a sort of password; you _should be **required** to enter it in-order to be allowed to connect._ That fact that you found a way to get it to work _without_ the API key isn't a bug, it is a security violation. It's like configuring a web service to allow log in knowing only the username, no password. It's not terrible, but it's also not a great position to be in. The system has security, it should be being used.

@crstian19. In your workaround, is the method of selecting OctoPrint as the host type a way to be allowed to enter the API key for securely connecting? In other words, is the security part still functional with your method?