supermerill/SuperSlicer #3345: Garbled slicer paths for both released version and nightly_dev build

kursatu, 2022-10-16:
### What happened?

I have been using nightly_dev build that I've made few weeks ago, in order to use Arachne.
Not sure if it is related or not but I've tried to change the GPU that it uses to on-board GPU rather than discrete GPU.
It somehow seems to have messed up a persistent state for both the released superslicer and nightly_dev version.
Now I'cant slice with neither of them. Prusa slicer works fine.
[image]
[image]

### Project file & How to reproduce

There is no reproduction steps. I am attaching a project from nightly_dev build.
The screenshots above are from the released Superslicer.
[Fuselage new.zip](https://github.com/supermerill/SuperSlicer/files/9795524/Fuselage.new.zip)

jhlchu, 2022-10-17:
You have fuzzy skin on. `Print Settings > Perimeter & Shell > Advanced > Fuzzy Skin = None` fixes it. If it fixes the issue, close.

kursatu, 2022-10-17:
The fuzzy skin setting was apparently causing it.

closed 2022-10-17 by kursatu