supermerill/SuperSlicer #3350: 'Ctrl + s' Should Save Changes to Print, Filament or Printer Settings

Garryb123, 2022-10-18:
I'm always frustrated when I have to make extra mouse moves and multiple mouse clicks just to save a change in Print, Filament or Printer Settings. All the mouse moves and clicks are un-necessary and the overhead quickly adds up to an unacceptable amount when making a lot of changes in these areas.

When on the Print Settings, Filament Settings or Printer Settings tabs please make 'Ctrl + s' save changes to whichever tab is active WITHOUT opening the Save Preset window.

Press 'CTRL + s' and changes would be saved to the profile. The job would be done in one shot just like all Windows programs do.

There doesn't seem to be any alternative method of saving changes to settings in these tabs.

jhlchu, 2022-10-18:
`Ctrl + S` is already bound to save the project file (*.3mf) regardless of whatever settings tab you are on so can't be used for this purpose. `Ctrl + S` behaves like any other Windows program, in that is save the current file.

Garryb123, 2022-10-18:
> regardless of whatever settings tab you are on

That is not the case for me. `Ctrl + S` within those tabs does nothing.

jhlchu, 2022-10-19:
> save the project file (*.3mf) regardless of whatever settings tab

It's saves to the project. There is no star* next to your project name in the title. It's the reason someone can open your project file and have the same settings as you had. You can unzip the 3mf to see it.