supermerill/SuperSlicer #3351: Avoid Crossing Perimeters Broken

Garryb123, 2022-10-18:
### What happened?

[image]
[image]
[image]

'Avoid Crossing Perimeters' is broken after a cetain layer.

I turned on 'Avoid Crossing Perimeters' and set the 'Max Detour Length' high so the nozzle would not cross the hole in this model.

I expected the nozzle would not cross the hole perimeters.

In this model 'Avoid Crossing Perimeters' works till layer 15 but is broken thereafter.

### Project file & How to reproduce

[NH_tangent_low.zip](https://github.com/supermerill/SuperSlicer/files/9805403/NH_tangent_low.zip)

### Version

Version 2.4

### Operating system

Windows 10 19044.2130

### Printer model

Custom printer with Duet controller

jhlchu, 2022-10-18:
There is no viable path to avoid crossing perimeters so it takes the shortest path. It can't go around the outside since your external walls are overlapping each other, going between them would count as crossing perimeter. Reduce your `Width & Flow > Perimeter = 0.25` and you will get the desired tool path (see image). It leaves just enough gap so it won't count as crossing periemter.
[image]

jhlchu, 2022-10-18:
If it fixes your issue, close issue.

Garryb123, 2022-10-19:
Thank you for taking time to explain.