supermerill/SuperSlicer #3352: Arachne Settings Missing

thecurtis76, 2022-10-18:
### Version
2.5.59.0

### Operating system type + version
MacOS 12.6 M1 arm

### Behavior
Arachne settings missing.

[image]

neophyl, 2022-10-18:
Duplicate of https://github.com/supermerill/SuperSlicer/issues/3094  
Please search for existing issues (both open and closed) before opening a new one.  The one linked is even on the first page still.

thecurtis76, 2022-10-18:
@neophyl My bad. I did a search, but, for some reason, I didn't see this one. Posting these is always a last resort for me, so I'll look harder next time. Thanks for pointing me in the right direction.

harriettubgirl, 2022-11-17:
Thanks for posting th is, I don't even see the other post neophyl is talking about. 

I'm having the same issue!