supermerill/SuperSlicer #3353: NetFabb repare service missing from Linux version?

Flip35, 2022-10-18:
### What happened?

I use both flavors of SuperSlicer (Windows or Linux) according to the PC I'm using.

On the Windows version, if I hover the mouse over the blue triangle icon on the left side of the file name, I have the option to repair the STL using NetFabb. This is not working on the Linux version.
[image]
[image]
[image]

### Project file & How to reproduce

[Projects.zip](https://github.com/supermerill/SuperSlicer/files/9808638/Projects.zip)

### Version

2.4.58.5

### Operating system

Linux Mint 20.3 (Ubuntu 20.04)

jhlchu, 2022-10-18:
> <h3 style="box-sizing: inherit; display: block; font-size: 24px; font-weight: 900; line-height: 1.25; margin: 48px 0px 16px; clear: both; color: rgb(76, 76, 76); font-family: Roboto, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">System Requirements for Netfabb 2023</h3><p style="box-sizing: inherit; margin: 0px 0px 24px; color: rgb(76, 76, 76); font-family: Roboto, sans-serif; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Netfabb is only available for Windows, not Mac.</p>

Operating System | Microsoft Windows 10 64-bit or 8.1 64-bit. If you run “Fusion 360 with Netfabb” then it will also run on Apple macOS Big Sur 11.x; Catalina 10.15; and Mojave v10.14.Note: 32-bit Windows, or older Windows platforms, as well as Linux or any OS by Apple, is not supported.
-- | --
CPU Type | Intel Core i5 (2.8+ GHz) with 8 cores
Memory | 64 GB of RAM
Display Card with OpenGL 3.3 support | 2 GB dedicated video card
Display Resolution | 1920 x 1080 minimum
Network Card | Required for network installation with FlexLM licensing
Java Runtime | 1.8u25 64-bit or newer, 32-bit Java is no longer supported.
System requirements for Local Simulation | Microsoft Windows 10 64-bit Microsoft Windows 8.1 64-bit Linux CentOS 8 64-bit Linux CentOS 7.4 64-bit Linux Red Hat 8 64-bit Linux Red Hat 7.4 64-bit

Looks like Mint isn't a supported distro.

neophyl, 2022-10-18:
Correct.  It is only available as part of a windows installation.  Its not that its missing from the linux version of SuSi, its that linux doesnt have the component used to access netfabb that the slicer needs.  Same as Prusa Slicer.

kozross, 2022-10-19:
Could you run under Wine to get the necessary stuff for this to work? Less optimal, but would be fine.

neophyl, 2022-10-20:
Does WINE have the full win10/11 operating system ?  
The whole reason netfabb works under certain versions of windows is an agreement between microsoft and autodesk allowing its OS to access it.  For example it doesn't work on windows XP.  It's part of the whole support windows 10 introduced for 3d models with its 3d builder system.

I would imagine it wasn't free either and that MS probably had to pay Autodesk for the privilege.  Sounds like something you need to lobby the linux team to add support for so that the slicer can access it on your chosen operating system.

kozross, 2022-10-20:
@neophyl I mean, Wine can and does emulate Windows 10. However, whether it emulates this specific thing or not I have no idea. FWIW, I did try running PrusaSlicer under Wine, but the feature didn't work there either. Whether it's because I didn't set something up correctly or whether it's outright impossible I have no idea.

neophyl, 2022-10-20:
Then I would guess that your wish to have netfabb access on slicers running on Linux is going to be out of luck.  Its not like they can code in functionality to the slicer either as its just making use of an OS provided 'hook' for it to send off the model to the Autodesk servers and receive back a 'fixed' version.

I thought there were various flavours of linux, surely some of those have teams that coordinate and develop them ?  However if Autodesk don't support any linux anyway then there doesn't seem much point to that either.

If wine is just an emulator and not a container with a proper OS install inside then I also suspect that it wont be a setting or configuration you can turn on either.  Might be better just running a virtual machine with a win 10 install inside it if you absolutely have to.  Sounds more trouble than just using Blender to fix any model errors though.

kozross, 2022-10-20:
@neophyl I've found a good middle ground. Netfabb for Linux does exist (in a much older version), and I can just manually open and fix my model in that. It's not ideal, but it's much easier than learning Blender (they say it has a learning cliff for a reason) or setting up a full VM.