supermerill/SuperSlicer #3354: Support Interface Issues

Durahl, 2022-10-18:
### What happened?

Greetings!

There's two issues I've come across concerning Support Interface Generation one of which I found an old entry about supposedly fixed which I guess had a glorious comeback after a mild setback? 🤨:

1. When selecting `Ironing` as the Pattern for the `Support Interface Layer` not only will it not apply the Ironing effect but it will also remove the second layer creating a one layer airgap kinda defeating its purpose:
[image]
Interestingly enough during the calculation it will ( occasionally ) for a split second have generated it properly but then waste it:
[image]

2. Selecting `Grid` from the `Support Material Style` creates a second layer on top of the first one going the same direction...
[image]
...instead of a 90° rotated one like how it is with `Snug`
[image]
Technically I wouldn't describe it as a HUGE deal but I found the removal of the Interface Layer in one piece a LOT less of a problem from it constantly tearing apart with the second layer being at a 90° thus keeping it together.

[image]

### Project file & How to reproduce

1. Load File.
2. Change `Support Interface Layer Pattern` to `Ironing` > 2nd Layer will not generate and neither will the Ironing.
3. Change `Support Interface Layer Style` to `Grid` > 2nd Layer will be at 0° Plebian Angle instead of  90° Master Race.
4. Flip Table ( depending on size of table a 2nd person may be required ).

### Version

2.5.59.0

### Operating system

Windows 11

### Printer model

Voron 2.4 with StealthBurner & ERCF

neophyl, 2022-10-19:
Sorry I cant find any attached file.  Project files need to be zipped.

Durahl, 2022-10-19:
Weird... Must have gone lost when I shuffled stuff around for readabilities sake 🤔
Added it to the end of the OP