supermerill/SuperSlicer #3355: SuperSlicer automatically-generated supports -- build-plate only -- failed to support the elbow in Michelangelo's David Sculpture

theschles, 2022-10-19:
### What happened?

Hi all, I downloaded https://www.thingiverse.com/thing:4660185 -- David's Michelangelo.  In SuperSlicer I told it to auto-generate supports, only from the build plate.  

Good supports were created -- except to support David's elbow.

Model:
[image]

Preview with just build-plate supports:
[image]

Preview focusing on the elbow and lack of support with build-plate supports:
[image]

After much experimentation, in order to get David's elbow supported was to de-select "Support on build plate only".  That added supports to the elbow...but increased material and print time considerably :(

Preview with supports everywhere:
[image]

Preview focusing on the elbow with supports everywhere:
[image]

Is there another configuration I could use that would allow me to keep supports as build-plate-only?

### Project file & How to reproduce

1. Download STL from https://www.thingiverse.com/thing:4660185 (or grab the project from here -- zip file with 3mf & stl is 52MB: https://www.dropbox.com/s/a31j79wxmz3kwgw/DavidStatue-tiltfix.zip?dl=0 )
2. Import STL into SuperSlicer
3. Configure auto-generated supports 
4. Configure overhang threshold of 0 (the default) for auto-detected overhang thresholds
5. Configure supports to be build-plate-only
6. Preview
7. See that elbow is _NOT_ supported (BAD)
8. Now configure supports to be everywhere
9. Preview
10. See that elbow is _NOW_ supported (GOOD BUT SLOWS THINGS DOWN AND WASTES MATERIAL)

Thank you in advance for your help!

### Version

2.4.58.5

### Operating system

MacOS 12.6

### Printer model

Sovol SV-01

neophyl, 2022-10-19:
The elbow is in no way above the build plate so of course supports are not generated when you select that option.  To support that requires supports on the base of the model.  If you check there are also other areas on the rear that need support but have none.  For the same reasons. 

The most optimal way is to define the areas you want supporting using the paint tool and then use the for support enforcers only option.

Also I'm getting warnings of empty layers when I slice your 3mf so id check the model. 
[DavidStatue-tiltfix_neo.zip](https://github.com/supermerill/SuperSlicer/files/9823587/DavidStatue-tiltfix_neo.zip)

theschles, 2022-10-19:
Hi @neophyl forgive me, I was confused.  My reference is Cura here -- and I thought that Cura had a way to build supports 100% from the build plate for any point needing support on a model.

Apparently not.  Apologies for bothering everybody.

closed 2022-10-19 by theschles

neophyl, 2022-10-20:
Cura does, it’s called tree supports. However it’s not a feature that is available in either Prusa Slicer or SuSi yet. It’s been requested though.

theschles, 2022-10-20:
Hi @neophyl actually I tried Cura 4 and Cura 5 last night just before my comment -- and David's elbow is still not supported with tree supports if one only builds supports from the build plate.  To get Cura to support David's elbow, one needs to build supports from everywhere.

Unless I'm doing something wrong in Cura...