supermerill/SuperSlicer #3357: Color blind adjustments

CadenjJaocbs, 2022-10-20:
I'm color blind and when it comes to looking at the speed view in the gcode screen, it can be very hard to see exactly what speed each feature is.  It would be amazing to have some sort of color blind adjustment for the 3 main types of color blindness; Deuteranomaly, Protanomaly, and Protanopia. 

This could be a feature in the printer tab or in the gcode screen that allows you toggle between each of the types and off.  All this could be is a color adjustment for the part itself, I don't see an issue when it comes to the rest of the software. 

If its not possible to have this its not the end of the world, but would be super nice to have.

jhlchu, 2022-10-21:
What OS are you using? Most OS have color filters integrated as accessibility features. Windows has a color filter built0in for deuteranopia RG, protanopia RG, and tritanopia BY.

CadenjJaocbs, 2022-10-22:
I did that and it helped but it can still be hard to see,