supermerill/SuperSlicer #3359: Better hiding of seams on cylinder-like rounded objects [Feature request]

laptop-laptop, 2022-10-22:
**Is your feature request related to a problem? Please describe.**

On PrusaSlicer 2.5, when printing elongated rounded object like a cylinder or bottle, all available options cannot hide the seams.
Random and Nearest seem to produce similar results where clusters of seams are spread on the rounded surface like fireworks on the sky. As for the other options, they just producd an ugly line of seams.

Does SuperSlicer have a better way to hide the seams?

**Describe the solution you'd like**

A solution that would allow better hiding of seams for this type of rounded objects.

jhlchu, 2022-10-22:
There isn't any way to hide seams on a rounded surface as it's entirely convex. There are no corners where seam could hide.
Alternatives:
1. Vase Mode
2. `Print Settings > Perimeters & Shell > Advanced > One-loop perimeter`

laptop-laptop, 2022-10-22:
Thanks for the suggestions. I need more than one perimeter so Vase Mode may be out of the question. Does one-loop perimeter print one perimeter also?

Can the Fuzzy skin feature somehow be modified to have ironing on the surface that could iron out the holes due to the seams?
Alternatively, is it possible to start and end each layer a bit inside the cylinder-type object so the opening is hidden inside the cylinder?

jhlchu, 2022-10-23:
> Does one-loop perimeter print one perimeter also?

Yes. Just apply the settings on your object to see.
> Can the Fuzzy skin feature somehow be modified to have ironing on the surface that could iron out the holes due to the seams?

If you have holes in your seam, adjust `Extra length on restart`, `Seam gap`, and `Wipe`. Just means you need to tune your seams.
> Alternatively, is it possible to start and end each layer a bit inside the cylinder-type object so the opening is hidden inside the cylinder?

Yes. `Printer Settings > Extruder 1 > Wipe Inside > At Start = True` and `Depth`.

laptop-laptop, 2022-10-25:
Thank you.

I do functional prints so one perimeter is not sufficient.

As for "Wipe Inside", shall I check both "At start" and "At end"? What depth value do you recommend?