supermerill/SuperSlicer #3360: FEATURE : Optimize travel moves

inspectionsbybob, 2022-10-22:
### Version
2.4.58.5

### Operating system type + version
ALL

### Behavior
As I watched my printer go back and forth (and back and forth and... ) building several things, Its movements were mostly back and forth over wide spaces just to put a tiny section down and then travel all the way back....  My Laser has an option to "optimize cuts & Travel," and this will do some simple (??? ha!) calculations to keep the head in one area until it is done in that area, rather than following the order the objects were created,  obviously at the cost of slicing time...

This Slicing option would do its best to do ALL of the printing in one "area" on a layer before moving on to the other areas.
This could GREATLY decrease print times on some prints, and could only help on the rest.. This also could reduce stringing if the printhead could finish MORE before invoking an extended travel/string.

There also could be an option for how far ahead to look in the code or how many recursions, as each step will slow down the slice.

_Is this a new feature request?_
YES

The laser software is MEERK40T and is also on Github.  The Optimization routines would be findable in Meerk40T's resources.
(For those also following this, MeerK40T is an open-source response to "LightBurn" which is big and expensive.  The software is still actively being updated, and it is very good if you are also a Laser Person)

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

user897943, 2022-10-28:
It is probably not as simple as you think - lasers remove material and the material has a constant surface. 3D printing adds material and has several layers differing in height. So it creates issues of adhesion, joins/seams, head crashing etc.. that would be my guess as to why it hasn't been further optimized. With 4D printing this would probably be possible since the print head can move in more directions than XY.

OldGuyMeltsPlastic, 2022-11-22:
On a plate with multiple objects/parts, it seems like each layer still moves from one part to the next, whereas maybe each layer could be analyzed (not a developer, not sure how this would work) to move from one blob of plastic to another blob of plastic in a more optimized fashion, rather than completing one object/part before moving on to the next.