supermerill/SuperSlicer #3371: E3D Spider Stresstest generating random lines

Arakon, 2022-10-25:
### What happened?

[image]
In the bottom half of the spider web, two straight lines are generated that should not be there. 
[image]

They're apparently created as plain perimeter lines.

Cura does not exhibit this behavior. Changing layer height doesn't seem to help, either.
Oddly enough, a different printer profile does not exhibit this behavior.

### Project file & How to reproduce

[StressSpider(1).zip](https://github.com/supermerill/SuperSlicer/files/9864183/StressSpider.1.zip)

### Version

Version 2.5.59 (latest nightly, September 13th)

### Operating system

Windows 10

### Printer model

VZBot

Pavulon87, 2022-10-26:
Looks like arachne bug. Sliced with a clasic perimeter generator without problems.
[image]

vovodroid, 2022-10-30:
@Arakon , try to increase Wall transition angle. In my case it helped.

vovodroid, 2022-11-06:
There is no this bug in **dev** branch, and in **nightly_dev** with **dev** merged.

By a way, what is function of this two branches? **nightly_dev** is considered more stable?