supermerill/SuperSlicer #3376: Top layer created despite "0" setting

vovodroid, 2022-10-27:
### What happened?

Despite top soild layers set to zero it's still created:

[image]

If Infill is set to 0 just external perimeter is printed:

[image]

### Project file & How to reproduce

[Merged.zip](https://github.com/supermerill/SuperSlicer/files/9876626/Merged.zip)

### Version

Version 2.4.58.5 (nightly_dev as well)

### Operating system

Wn10 x64

### Printer model

Biqu B1 SE

neophyl, 2022-10-27:
Its not top infill really its internal infill.  You are changing the infill % and the amount changes.  Even on the key when sliced its labelled up as internal infill.  
It is because you have Print Settings>Perimeters & Shell>Quality> Only One perimeter > On Top Surfaces enabled.  The lower layers have sufficient perimeter count to preclude the use of any infill.  On the top because of that option there is only one perimeter allowed so it has room for infill, which is what it is doing.  Infill NOT topfill.

No issue here, its just doing what you have configured.

vovodroid, 2022-10-27:
Indeed. Superslicer has quite sophisticated configuration)))
Thanks!

closed 2022-10-27 by vovodroid