supermerill/SuperSlicer #3379: First layer bed temp ignored

dJOS1475, 2022-10-28:
**Bug description**
On 2.5.59.0 the first layer bed temp is being ignored and the “other” temp is being used from layer 1 onwards

**Expected behavior**
I would expect the first layer to be printed at the specified temp.

**To Reproduce**
Steps to reproduce the behavior:
1/ Change first layer bed temp to something different to “other”.
2/ print model

**Desktop (please complete the following information):**
 - OS: MacOS Intel
 - Version [e.g. 2.5.59.0], 
 - Printer model (any)

diggit, 2022-10-28:
Can you share your Printer -> Start G-code?

dJOS1475, 2022-10-28:
Sure thing:

```
; Klipper Config
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode

M117 Homing X/Y ...
G28 ;home all axis
G28 X0 Y0 ;move X/Y to min endstops
G28 X Y ;move extruder back to front left corner

M107 ;start with the fan off
M117 Bed Heating ...

; Section to heat bed
M190 S[bed_temperature] ;wait for bed temperature
M117 Bed Settling ...
G4 P60000 ; Wait 1 minute to let bed temps settle
M400 ; Wait for delay

M117 Auto Gantry Level ...
G28 ;home all axis
Z_TILT_ADJUST ; Z Axis Level

M117 Auto Bed Level ...
G29 ; Auto Bed Level

G28 X0 Y0 ;move X/Y to min endstops
G28 X Y ;move extruder back to front left corner
G1 Z15.0 F[travel_speed] ;move the platform down 15mm

; Section to heat hot end
M117 Ext Heating ...
M109 T0 S[temperature_0] ;wait for hotend temperature

M117 Start cleaning ...
G92 E0 ;zero the extruded length
G1 F200 E10 ;extrude 10mm of feed stock
G92 E0 ;zero the extruded length again

M117 Intro line ...
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X2.0 Y20 Z[layer_height] F5000.0 ; Move to start position
G1 X2.0 Y100.0 Z[layer_height] F1500.0 E15 ; Draw the first line
G1 X2.4 Y100.0 Z[layer_height] F5000.0 ; Move to side a little
G1 X2.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line

G92 E0 ; Reset Extruder
G1 E-1 F500 ; Retract filiment by 1 mm
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X6 Y20 Z0.3 F[travel_speed] ; Move over to prevent blob squish
G1 F[travel_speed]
G92 E0 ; Reset Extruder
M117 Printing...
```

diggit, 2022-10-28:
well you are explicitly heating up bed to default (non 1st layer) temperature
`M190 S[bed_temperature] ;wait for bed temperature`

Try:
`M190 S[first_layer_bed_temperature];`

Fun fact: I had is same way you have so I had to fix it too...

Consider similar thing for M109 (hotend temperatures) with: `first_layer_temperature[initial_extruder]`

dJOS1475, 2022-10-28:
Ah, that'll do it, thank you - I'll make the change.

dJOS1475, 2022-10-28:
and yes that worked - thank you.

I haven't tried the HE temp but I'll do that later.

diggit, 2022-10-28:
If you think, that issue was resolved, feel free to close it :)

dJOS1475, 2022-10-28:
This issue is resolved

closed 2022-10-28 by dJOS1475