supermerill/SuperSlicer #3380: Default Fan speed ignored

adripo, 2022-10-28:
### What happened?

`min_fan_speed` is set at 35%, but gcode is created starting from 0%

[image]

[image]

### Project file & How to reproduce

Any project behave the same.

Filament settings -> Default fan speed: 35%
Gcode Preview -> View Fan speed -> Values below default fan speed

### Version

2.4.58.5

### Operating system

windows 10

adripo, 2022-10-28:
I would expect to have:
- 1 layer: 0%
- 2 layer: 35%
- 3 layer: ~60%
- 4 layer: 100%

[image]

Am I missing something?

diggit, 2022-10-28:
"Full fan speed at layer" is definitely confusing. It means, that at layer 4, this limiter will stop limiting and fan will be on its target speed. Thus:
- layer 1: 0%
- layer 2: 12%
- layer 3: 23%
- layer 4: 35 %
I am unsure, why 5th layer gets to 80~90 %. Too short layer time I guess?

adripo, 2022-10-30:
What is the purpose of default fan speed then? It doesn't affect the "minimum PWM that my fan needs to work" as explained in the details. I also checked `Run the fan at default speed when possible` but no change.

legend069, 2022-11-29:
default fan speed is used for extrusion roles that currently don't have a fan sped override set