supermerill/SuperSlicer #3383: Link for "new release version 2.4.58.5 is available!" links to Prusa, not SS on Github.

darrylhadfield, 2022-10-28:
### What happened?

When clicking on "See Download Page" it sends me to a link on prusa3d.com and does a quick redirect, ending up at https://www.prusa3d.com/page/prusaslicer_424/ .

### Project file & How to reproduce

When I open up SuperSlicer, thet main display window shows a box at the lower right that reads, "New release version 2.4.58.5 is availble.  See Download page."

Where "see download page" links to https://www.prusa3d.com/slicerweb&lng=en_US
which in turn redirects to https://www.prusa3d.com/en/page/prusaslicer_424/
which in turn redirects to https://www.prusa3d.com/page/prusaslicer_424/ .

### Version

2.4.58.4

### Operating system

Windows 10

### Printer model

n/a

doctor-bob, 2022-10-29:
I have a feeling that something strange is going on with superslicer....it seems like nothing is going on, or something behind the scenes...?

neophyl, 2022-10-29:
Already reported several times, been fixed in later versions. When merged from PS it slipped through and was still pointing to the original address.

darrylhadfield, 2022-10-29:
> Already reported several times, been fixed in later versions. When merged from PS it slipped through and was still pointing to the original address.

I searched first, and couldn't find anything when looking for "upgrade", or "prusa", or "new version".

Nothing within the last month.

not sure what you mean by "fixed in later versions"...  I was running the next-to-latest version.

A check of the bugfix notes shows #3069 but that was reported back on August 13 and was only fixed for the latest version - the issue was still there in 2.4.58.4, which is what I was running.

Closing the issue as it looks like it's been fixed.

closed 2022-10-29 by darrylhadfield

neophyl, 2022-10-29:
https://github.com/supermerill/SuperSlicer/issues/3262  And it’s also been in the discussions section. 

There are the unstable versions based on ps 2.5 that are available too.  SuSi releases are a bit all over the place at the moment so you can’t just go by the version number.