supermerill/SuperSlicer #3384: Arachne produces connecting lines in the air

vovodroid, 2022-10-29:
### What happened?

When perimeters count > 1 strange lines are produced:

[image]

[Nozzle-Fillet001.zip](https://github.com/supermerill/SuperSlicer/files/9894371/Nozzle-Fillet001.zip)

Probably they are supposed to be travel movements?

### Project file & How to reproduce

Slice attached project

### Version

Latest night_dev f8c9106c

### Operating system

Win10x64

### Printer model

Biqu B1 SE

neophyl, 2022-10-29:
Looks similar/same as https://github.com/supermerill/SuperSlicer/issues/3371

vovodroid, 2022-10-29:
For what it's worth, but increasing Wall transition angle fix it at 33 degrees (first left line, and then right).

ChaosBlades, 2022-11-01:
There are a lot of issues with Arachne in SuperSlicer that would be resolved by merging the latest code from PrusaSlicer. If I remember correctly the version of Arachne currently in SuperSlicer is from the PrusaSlicer Arachne Beta not the stable release.

vovodroid, 2022-11-01:
> would be resolved by merging the latest code from PrusaSlicer

Not sure. On September 13 https://github.com/supermerill/SuperSlicer/tree/nightly_dev was updated to latest fixes from Prusa. I also just copy latest Arachne files from Prusa with some files from Filles and Geometry, nothing changed. Probably Prusa has another default config values.

ChaosBlades, 2022-11-01:
Try slicing it in Prusa Slicer. If you get the same result then it might be best to upstream this issue to PrusaSlicer to fix. Also might be a good idea to check on a clean install of SuperSlicer to try and narrow down if a specific setting that you changed being enabled/disabled is causing the issue.

I can't even slice your project file. I get 'No extrusions were generated for objects.' error on 2.5.59. Probably something that exists in the latest dev that my version does not have.
If I use my own profiles with variable layer height I don't have any issues.

neophyl, 2022-11-01:
There is already at least one and I think more of these issues on the PS github regarding extraneous lines with Arachne.  Often adjusting the settings helps.

vovodroid, 2022-11-01:
> Try slicing it in Prusa Slicer.

I tried, it doesn't happen, but probably due to different settings, not due to code fixes.

> might be a good idea to check on a clean install of SuperSlicer

You don't need clean install, just choose built in profiles. It happens with original MK3 profiles as well.

leezus-nz, 2022-11-06:
I've had this issue too. Fixed it by rotating the part 180 deg. ¿?