supermerill/SuperSlicer #3385: Only Getting 1 Top Layer When 6 Are Specified

CCS86, 2022-10-29:
### What happened?

Hey guys, can anyone see why layer 94-97 are getting no top infill, when I have 6 top layers specified?

[image]

### Project file & How to reproduce

[bigger-helical-gears.3mf.txt](https://github.com/supermerill/SuperSlicer/files/9894374/bigger-helical-gears.3mf.txt)

### Version

2.5.59.0

### Operating system

Windows 10 x64

Pavulon87, 2022-10-29:
I'm not sure why(probably bug), but if you check "Ensure vertical shell thickness" then it works fine.

CCS86, 2022-10-29:
Awesome, thanks man! I forgot about that check box